riscvarchive/riscv-v-spec #1: Improve rounding mode names

jhauser-us, 2018-07-27:
For one of the fixed-point rounding modes, you have
> `rup` round-up (+0.5 LSB)

I suspect you mean round to nearest, with ties rounded up.  That would be more properly called "round nearest-up", not "round-up".

For better consistency, "truncate" is also known as "round down", and "jam" is "round to odd".  If you've never heard of "round to odd", see for example:
https://www.exploringbinary.com/gcc-avoids-double-rounding-errors-with-round-to-odd/
https://developer.arm.com/docs/dui0801/g/a64-simd-scalar-instructions/fcvtxn-scalar

aswaterman, 2018-07-27:
FWIW, the F-extension chapter uses `rup` to mean "round towards +inf"

kasanovic, 2019-01-07:
Adopted John's names.

closed 2019-01-07 by kasanovic