riscvarchive/riscv-v-spec #2: Integer SGE and SGEU are just as redundant as SNE

jhauser-us, 2018-07-30:
In the subsection "Vector integer arithmetic instructions",  it's noted that
> SNE not needed with complementing masks

For integer comparisons, the same observation is equally true of SGE and SGEU; they are the complements of SLT and SLTU respectively.

brucehoult, 2018-10-12:
Section 4.3 lists
VSEQ, VSGE, VSLT, VSNE
VFSEQ,VFSGE,VFSLT,VFSNE

5.2 lists
VSEQ, VSLT, VSLTU, VSGE, VSGEU

9.5 lists
VFEQ,VFLT,VFLE

I guess there are four things here:

1) section 9.5 loses the "S", although these seem clearly the same operations called VFSEQ,VFSGE,VFSLT,VFSNE in section 4.3

2) Given .t and .f predication, EQ and NE are equivalent. Though maybe for other uses having both could save having to do a VXORI 0x01

3) LT and GE are redundant given that you can swap registers

4) in section 9.5 LT and LE are inconsistent with LT and GE elsewhere

aswaterman, 2018-10-13:
@brucehoult LT and GE aren’t made equivalent by swapping the registers - that changes LT to GT. Supplying both LT and GE is desirable.

I suggest keeping both SEQ and SNE, because sometimes the predicate results feed into things other than predicate inputs, and so it can save the VXORI.

brucehoult, 2018-10-17:
Closed? On what basis?

The inconsistency in calling the same operations VFSEQ,VFSGE,VFSLT,VFSNE in one place and VFEQ,VFLT,VFLE in another place has not been corrected.

The inconsistency of listing VSLTU and VSGEU in section 6.2 but not in the instruction encoding still exists.

commit 63c5cc43 (2018-10-18): Make comparisons consistent, internally and with scalar ISA

Fixes #2
closed 2018-10-18 by colinschmidt