riscvarchive/riscv-v-spec #5: vemaxw for 128 bit

aka-sps, 2018-10-12:
2.1.2. vemaxw[5:3] is not define 128 bit values (rv128, long long double, etc).

colinschmidt, 2018-10-16:
128 bit values are currently propsed to not be supported in the base.
The extended (6 bit) encoding for vemaxw would contain the larger bit widths.

closed 2018-10-16 by colinschmidt

aka-sps, 2018-10-16:
128-bit alignment is implicitly reserved in architecture (for example, 16-byte stack operation in RVC). 
128 bits is the natural alignment of RISC-V ISA and the possible size of the operand, for example, in cryptographic ISA extension.
It is possible that the vector extension architecture will not be able to support the entire instruction basis for 128 bits, but encoding of 128 bit elements should be reserved initially to avoid future problems.

colinschmidt, 2018-10-16:
128 bit element widths are reserved currently, and usable with VTypes
The cryptographic ISA extension should mandate VTypes.

Furthermore, there is not encoding space for explicitly typed 128bit opcodes so you would be required to use the polymorphic variant which is enabled with VTypes.

aaronfranke, 2019-04-16:
Can someone clarify, does this mean that support for 128-bit values such as quadruple-precision floats is dependent on additional vector extensions such as RV128 in addition to the base vector extensions?