riscvarchive/riscv-v-spec #7: Configurations for floating point calculations cannot use byte registers.

aka-sps, 2018-10-12:
From "9.1. Vector minimum element width" vemaxw for any Floating-Point operation should be not less than 16 bit word, so any byte vector registers can not be configured.

Is it right?

colinschmidt, 2018-10-16:
The only limitation in the base encoding is being unable to use bytes and double precision floating point registers, singles precision and bytes work fine.
The extended encoding(vtypes) supports arbitrary power of two element widths.

If you are using bytes and octowords, the number of octowords will dominate the vector length multiplier you would be able to get. 
For example if `vemaxw` = 8bit, the minimum configuration with 1 octoword and 2 singlewords you would only get a vector length multiplier of 3.
Whereas, if you use  `vemaxw` = 16bit then with 1 quadword, 1 doubleword, and 2 singlewords you can get a vector length multiplier of 4.

In addition, if you care about memory traffic you can still use byte loads into the wider 16bit registers.
Finally, adding an octoword field to the base `vregcfg` would introduce extra bits which has a non-trivial cost given the current encoding just fits in the vector immediate size.

aka-sps, 2018-10-16:
Please provide the natural reasons for these artificial constraints.
Perhaps the base encoding is incorrectly defined?

Byte elements are the smallest size for flags and small indices.
Double precision elements are determined by the algorithm itself. For the best use of the register file and for the maximum length of the processed vectors, the use of vector registers with the elements of the most suitable size is required.

In my opinion, the allocation of CSR bits to the detriment of the main functions is premature.

aka-sps, 2018-10-16:
Closing of issue is unconfirmed.

From spec (lines 920-923):
### Vector minimum element width
> An illegal instruction exception is raised when trying to execute a vector floating-point instruction for a precision that does not fit in the current `vemaxw`.

Verbatim: `vemaxw` should be encoded as `100` (64-bit) to use instruction `vfadd.d` and therefore all elements of all vector registers should be 64-bit. This is true?

Or, above claim should be rewritten as

> An illegal instruction exception is raised when trying to execute a vector floating-point instruction for a precision that does not fit in the vector element.

Please, confirm 1st or 2nd

colinschmidt, 2018-10-17:
It should be rewritten as the second.
`vfadd.d` Is a double precision addition.
An illegal instruction exception should be raised based on the operations table. In the current proposal any non-matching source and destination register widths will raise an illegal instruction exception.

commit fc8d06a4 (2018-10-18): Correct fp illegal inst on element width

Fixes #7
closed 2018-10-18 by colinschmidt