riscvarchive/riscv-v-spec #8: Vector config doesn't cover all useful cases

brucehoult, 2018-10-12:
It would be useful with (for example) a 32 register config to be able to have more than 7 double size registers or three quad size registers.

Logically, given that v0 must remain available in single size as a mask, it might be useful to have up to 15 double size registers (leaving 2 single size), or up to 7 quad size registers (leaving 4 single size, or 2 single size and 1 double size).

Adding 1 bit to each of the fields specifying the number of double and quad registers would achieve this, making 9 bits in total (along with the one reserved for dynamic typing).

Even if this does not fit into an 8 bit immediate field in a proposed "CONFIG" instruction (not present in the current encoding proposal), the full set could still be accessed by using a Load Immediate followed by a Move TO CSR.

There are actually 12 bits available in a Load Immediate instruction, which would allow room for a completely non-overlapping representation:

5 bits vregmax
4 bits number of d registers
3 bits number of q registers

12 total. Perfect :-)

brucehoult, 2018-10-12:
No, that doesn't quite work, because you've lost the reserved bit vregcfg[7] (intended for vtypeen?)

If the bits are taken from the hi bits in the Move to CSR then a LUI could be used to load the configuration instead of a LI.

a0u, 2018-10-18:
With 24d41e2adbd1557fbca10a42a6a068b140a46987, I've reworked the `vregcfg` CSR field to use a 12-bit encoding that compresses to the previous 7-bit encoding for the `vconfig` immediate.

As you note, allowing `vregmax` to vary between 1 and 32 inclusive makes the full configuration space expressible, which is why I chose this scheme as an initial design point for debate.  However, it's not without some complexity cost, as the alignment constraints (if retained) can lead to some awkward fragmentation in the register mapping.

For example, with `vregmax = 11`, `q = 1`, `d = 2`:
```
v0 v1 v2    v4             v8     v10
 s  s  d  -  q  -   -   -   d   -  s
```

Another option is to quantize `vregmax` like before (32, 16, 10, 8 ...), which enables a 4-bit encoding for `vregmax` and simpler conversion from `vfastcfg` in terms of logic:

| `vregmax` | CSR Encoding | `vfastcfg` Prefix|CSR->Prefix|
|---|---|---|---|
32|1110|11|`vregmax[2:1]`
16|1100|10| `vregmax[2:1]`
10|1011|011|`vregmax[2:0]`
8|1010|010|`vregmax[2:0]`
6|0110|0011|`{1'b0, vregmax[3:1]}`
5|0100|0010|`{1'b0, vregmax[3:1]}`
4|0011|00011|`{1'b0, vregmax[3:0]}`
3|0010|00010|`{1'b0, vregmax[3:0]}`
2|0001|00001|`{1'b0, vregmax[3:0]}`
1|0000|00000|`{1'b0, vregmax[3:0]}`

However, the bit pattern is less intuitive, and the sparseness makes it more difficult to allocate the unused encodings for other purposes in the future.

----
Taking a step back, I am not entirely content with how the `vregcfg` scheme is "mechanically" dissimilar to the original `vtype<n>` proposal, which we prefer to retain for the polymorphic extension.  It is clearly possible to map the `vregcfg` configuration space to the `vtype` space, but the two are rather different ways to express that information.  If a moderate increase of CSR state can be tolerated, it seems more elegant for the configuration scheme in the base vector extension to be a subset of vtypes: i.e., 2-bit `vew` field per vector register.

brucehoult, 2018-10-18:
As remarked on the call last week, it's easy enough for a compiler to deal with that.

    #include <stdio.h>
    #include <stdlib.h>
    
    void alloc(char *regs, int vregmax, int q, int d){
        for (int i=0; i<=vregmax; ++i) regs[i] = 0;
        int count[3] = {vregmax-2*d-4*q, d, q};
        char letter[3] = {'s','d','q'};
        for (int scale=2; scale>=0; --scale){
            int sz = 1<<scale;
            int pos = vregmax & ~(sz-1);
            for (int n=0; n<count[scale];){
                pos -= sz;
                if (scale ? pos<=0 : pos<0){
                    printf("Illegal config\n");
                    exit(1);
                }
                if (regs[pos]) continue;
                for (int i=0; i<sz; ++i) regs[pos+i] = i ? '-' : letter[scale];
                ++n;
            }
        }
    }
    
    int main(int argc, char **argv){
        if (argc<4){
            printf("Usage %s vregmax q d\n", argv[0]);
            return 1;
        }
        char regs[33];
        alloc(regs, atoi(argv[1]), atoi(argv[2]), atoi(argv[3]));
        printf("%s\n", regs);
        return 0;
    }

Running...

    bruce@nuc:~/programs$ gcc vreg_alloc.cpp -o vreg_alloc
    bruce@nuc:~/programs$ ./vreg_alloc 11 1 2
    ssd-q---d-s
    bruce@nuc:~/programs$ ./vreg_alloc 11 2 1
    Illegal config
    bruce@nuc:~/programs$ ./vreg_alloc 32 3 2
    ssssssssssssssssd-d-q---q---q---

However I don't know how easily hardware can do this calculation.

brucehoult, 2018-10-18:
Well, I guess the fast way for hardware to do this is with a lookup table. Twelve bits of config gives 4096 entries (of which 803 are valid). The output will want to be 2 bits for each of the 32 vreg numbers specifying whether that register number is invalid, single length, double length, or quad length. So that's 64 bits, 8 bytes per entry. Total 32 KB of table. I expect with a bit of cleverness (e.g. two-levels of table) that could be halved.

Is that too much? Is it worth making it a CAM to have 5x fewer entries?

grnewell, 2018-10-29:
It seems this and issue #44 are on the same topic

kasanovic, 2018-12-01:
Closing this issue as it no longer applies to current design.

closed 2018-12-01 by kasanovic

kasanovic, 2018-12-02:
> @kasanovic <https://github.com/kasanovic> There is now no way to indicate you won't be using all 32 vector registers?

I just pushed an update that describes the vlmul field to trade architectural vector registers for longer vectors.

Krste

brucehoult, 2018-12-02:
@kasanovic cool! Looks good.

It seems that it can be optional for hardware to implement vlmul. If the program's setvli specifies vlmul = 0b11 for example, then a correct program must only use v0..v3. However hardware could ignore this completely and use only standard length vectors and everything will work fine, if suboptimally.

aswaterman, 2018-12-02:
For everything to "work fine" with vlmul hardwired to 0, you'd need an additional ABI constraint.  Since the regfile organization is now exposed, it's possible to set vlmul=3 and do some computation on the lower-numbered registers; then set vlmul=0 and observe the changes in the higher-numbered registers.  If vlmul is hardwired to 0, that obviously won't have the same effect.

brucehoult, 2018-12-02:
I think that's an example of unnecessary leaking of the the (an?) implementation into the spec.

The spec should just say that after changes of vlmul the hi registers have unpredictable contents.

Punning the contents of individual registers is already bad enough!

kasanovic, 2018-12-02:
We're trying to keep implementation-specific unpredictable state out
of unprivileged ISA.

The punning helps with opaque save/restore, where code
saving/restoring state has no idea of what's in registers.

Krste

| I think that's an example of unnecessary leaking of the the (an?)
| implementation into the spec.

| The spec should just say that after changes of vlmul the hi registers have
| unpredictable contents.

| Punning the contents of individual registers is already bad enough!

| —
|

brucehoult, 2018-12-03:
@kasanovic I agree that opaque save/restore is very desirable, however the save/restore code does not have to understand what it is saving or restoring. It only needs a guarantee that if it puts all the bits back exactly the way it found them then things will work.

i.e. it needs to be punned (not zeroed), but the details of the punning can be undefined in the ISA.

kasanovic, 2018-12-03:
From experience, verification is a major pain if you have implementation-defined state.
It seriously complicates golden model development and/or random test generation.

There’s also the problem of latent bugs in shipping software.

Krste