riscvarchive/riscv-v-spec #12: Shift counter value interpretation

go-sc, 2018-10-16:
1. Is shift amount argument going to be signed or unsigned? How do we interpret its value if it is negative?
2. What if shift amount is larger or equal to element width? The resulting value in that case may be set to some constant like -1 or 0. Or, like in base ISA, shift amount is taken modulo element's width?

colinschmidt, 2018-10-16:
The shift amount value interpretation will be the same as the RISC-V scalar ISA.

Explicitly, the shift amount is unsigned, the shift amount will be module the register's element width.

aka-sps, 2018-10-16:
From specification (line 468)
> Source operands can be of different widths, but with a difference of no more than a single factor of 2

Which vector registers width (`src1`, `src2` or `dst`) is a modulo?

Please, reopen issue.

colinschmidt, 2018-10-16:
As in the scalar ISA:
> In RV64I, only the low 6 bits of rs2 are considered for the shift amount.

The lower log2(`src1` width) bits of `rs2` will be used.

aka-sps, 2018-10-17:
Conclusions:

- maximum number of shifts is determined by the XLEN of the base platform, but not by the vector element size
    - RV32V cannot shift 64-bit elements
    - The same code on the same configuration for the same data will produce different results on platforms with different XLENs.

go-sc, 2018-10-17:
This issue can be closed when the spec is updated.

commit 831e766d (2018-10-18): Add shift explanation

Fixes #12
closed 2018-10-18 by colinschmidt