riscvarchive/riscv-v-spec #13: Masked elements zeroing motivation

go-sc, 2018-10-17:
Section 5:
> Instructions writing vector shapes, write zero to the destination elements that are masked off.

This needs more reasoning. Why don't just leave masked elements unchanged? Why zero?
A user can easily fill masked elements with any needed value (not necessarily 0) using mask complement.

aka-sps, 2018-10-17:
Why is `0` selected as a filling (neutral) element?
Why, for example, is not sequence 1, for initialization by the value of qNaN?

aswaterman, 2018-10-17:
It’s to simplify the implementation of register renaming. If you don’t write the entire vector, you need to re-read the original vector to copy the unmodified values to the new physical register.

This should be explained in the commentary.

@aka-sps Producing the value 0 instead of -1 will occasionally shorten some handwritten code sequences, since it’s the additive identity.

aka-sps, 2018-10-18:
@aswaterman 
`Add` operation  is worst case (carry bits) in comparison with bitwise operations, especially for register elements of `d` and `q`  size, while bitwise operations can be executed for register elements parts independently.

Handwritten code can use `and` identity for `-1` or `or` identity for `0` too.

- [ ] The basis of `identity` should be discussed together with the motivation of using NaN-boxing.

aswaterman, 2018-10-18:
I do agree that the chosen value should be carefully considered. Zero remains the obvious choice for integers, but the canonical qNaN is also a defensible choice for floats.

vbx-glemieux, 2018-10-18:
Writing 0s to masked elements is an optimization to simplify OOO microarchitectures. It does not belong in the vector spec, and I'll explain why below.

We used to think delay slots were a good idea -- a microarchitectural implementation issue creeped into the ISA definition. We now believe that branch prediction is better and handles the case of differing pipeline lengths more elegantly. The lesson learned from this was that microarchitectural features should not be exposed as optimizations in the instruction side effects.

In the same way, writing 0s might make OOO implementations simpler, but it also emburdens ALL implementations with a more restrictive implementation requirement that might preclude other forms of microarchitectural optimizations. If this sort of decision is being made, it needs to be justified with quantitative analysis, not with arm waving.

Here is my full argument:

(1) Writing 0s is counter intuitive to hand-coding and to compiler writers.

the unmasked version:
vaddi v1, v1, 1  // increments all elements

the masked equivalent:
vaddi v1, v1, 1, v0.t // increments all elements where v0<i>.t==1, zeros other elements

People learning the ISA will expect the masked vaddi to only modify some elements of v1, not to modify all of them. But this is not so. I'm sure this will be the future source of aggravation for many novice programmers debugging their code.

Instead, the following code sequence is required to restore the expected behaviour:

vaddi v2, v1, 1, v0.t
vmerge v1, v2, v1, v0.t

This new sequence is longer (slower), and has the side-effect of using an additional vector register v2 (which may slow some code down due to extra spilling).

The advantage of being able to execute vaddi out-of-order is partially negated by the dependence of the vmerge on vaddi.

Macro-op fusion can't fix this, because it needs to write to both v1 and v2 as side effects.

This is a bit of a problem for compilers that want to use the fastest code sequence possible. In the case where v0<i>.LSB == 1 for all i, the compiler would want to use the unmasked vaddi sequence because it is shorter; detecting this case adds a lot of effort to the analysis.

(2) @aswaterman claims "the vast majority of operations are unconditional". This is untrue. Any code that has an early exit feature, from computing fractals to doing viola-jones / adaboost style object detection, will have the entire inner loop being masked operations.

(3) @aswaterman claims "or have the mask bit set". This is just conjecture and is unproven. Again, code with an early exit feature starts by computing data on all elements, and gradually removes elements by adding them to the mask as they exit early. At the end of the loop, the vast majority of elements have the mask bit cleared, not set. And this is for long sequences of masked instructions (see (2) above).

(4) Writing 0s past VL and to elements that should be untouched takes effort to optimize in the microarchitecture. Likewise, making a microarchitectural optimization that effectively writes a scalar result to all elements, but only writes to one element, takes effort. These decisions make *all* vector implementations more complicated -- making it impossible or inefficient to create simple vector implementations. How should a simple implementation write 0s all the way up to maxvl? The simplest way is to take the maximum time to run, no matter what the value of vl. This is a huge performance hit on simple implementations. Instead, implementers are forced to create a gang of flip-flops that are to be used to indicate that a register holds a value of 0, and another gang to indicate a register replicates element 0, then extra mux inputs are needed to select either 0 or element 0 in the case these registers are set. For a simple implementation with 32 vector registers of 128 words, this requires 16384 individual flip-flops. If users want to create a vector implementation with a small number of lanes (even one lane can give 10x speedup on things like FIR), this is a huge overhead. Writing 0s is only an optimization to make OOO implementations *easier*, and by doing so *all* implementations are forced to pay a heavy price.

(5) Alternatives have not been discussed. For example, an OOO implementation could use macro-op fusion on the following code sequence:

vxor v2, v2, v2
vaddi v2, v1, 1, v0.t

It could then take this as a "hint" that v2 is to be written with 0s, and merge the 2 instructions into a single one. This does make OOO implementations more complex, but face it they are already complex and paying an area penalty, and at least it doesn't emburden all simple implementations with an counterintuitive requirement.

(6) No quantitative analysis has been done to show that writing 0s followed by a vmerge instruction is actually faster than doing the intuitive thing. In this era, I thought the whole premise of RISC was to ensure only the minimal definition was included, in order to make the common case fast, and ideally it would be justified with benchmarks. Are there any code sequences that justify the current decision to write 0s? These should be shared as part of the justification.

(7) Forcing implementations to write 0s makes it very difficult to implement other techniques to skip over masked computations entirely, like density time masking. Thus, a decision is being made to favour one microarchitectural feature (OOO) over another (density time masking).

aka-sps, 2018-10-18:
@aswaterman 
I agree with you that we should avoid the red herring.
For example, the premature and mandatory determination of the values as zero for unevaluated elements.
Or, for example, we should avoid encoding 376992 useful configurations by 8-bit values.

You offer your own kind of micro-architectural solution, based on renaming vector registers entirely, and not, for example, on renaming elements of these registers or on renaming slices/chunks of these vector registers.
And you can do it.

Unfortunately, in real tasks, apart from primitive elementwise operations, the number of elements requiring zeroing can be significant. We analyzed a number of algorithms for linear algebra, sorting, searching, Fourier.
For many of them, the power of the set of processed elements is less than "MAXVL".
So, element zeroing (masked or after `vl` index) can be a mass operation.
I (and the battery of my device) do not want this.
And I want to do it.

Unnecessary zeroing is not only energy, but also time.

There are two approaches to masked operations.

1. Functional approach calculates result for vector elements subset as vector of calculated values and does not involve the calculation of other elements.  An explicit merge is required.
2. Object-oriented approach is similar [std::mask_array](https://en.cppreference.com/w/cpp/numeric/valarray/mask_array). An approach to renaming the entire register for this method is not suitable.

We should not resolve this issue without discussion and the views of other developers.

Proposed mandatory zeroing is Procrustean bed for future microarchitectures.

My simplest proposal: *Let's leave non-calculated elements undefined.*

If a common solution cannot be reached, then this question should be put to vote.

vbx-glemieux, 2018-10-18:
I don't fully accept the argument that writing 0s is necessary to make OOO implementations better.

Big OOO cores will presumably have 3 register file read ports in favour of fused multiply-add instructions.

Thus, for the vast majority of 2-operand instructions, they already have the ability to use the 3rd port to read out the previous destination register contents when implementing masked operations:

vadd v3, v1, v2, v0.t

Yes, a fused multiply-add that uses masking will need a 4th read port. Hopefully that is an uncommon case. Or, perhaps it is sufficient to avoid executing such instructions OOO (and to use the same physical register as the destination without renaming, making the 4th read unnecessary).

vbx-glemieux, 2018-10-18:
> My simplest proposal: _Let's leave non-calculated elements undefined._

@aka-sps This does make the implementation simpler. However, it is less useful because a vmerge will be required. In the cases above where I discuss exit-early algorithms, this will slow down the inner loop because the elements that have been fully computed already must remain untouched, not turned into undefined values. The overhead this adds depends upon the precise algorithm, and in the case of a long sequence (like the fractal or adaboost examples), it might only add 1 instruction after a sequence of 10 or so instructions. But in other cases, with tighter inner loops, the result will be more dramatic. Also, since it requires operations on all elements (both masked and unmasked elements), the merge is slower than the other instructions.

FYI, I have some first hand experience writing vector code that uses masking and gets speedups from density time masking. Writing 0s, is horrible. Leaving values undefined is slightly better, but I still don't agree when it is equally easy to leave the original values untouched, even on OOO implementations.

colinschmidt, 2018-10-29:
This was discussed in the vector working group meeting on 10/16/2018 (see slides for detailed pros and cons).
The decision was to return to the original proposal to leave elements in place, rather than zero.

commit 57264d20 (2018-10-29): Change masked and off-the-end behavior to unmodified

This fixes #13
This was discussed and decided at the WG meeting on 10/26/2018
closed 2018-10-29 by colinschmidt