riscvarchive/riscv-v-spec #14: Do not write 0 after vl

go-sc, 2018-10-17:
Section 5:
> The active vector length in `vl` determines the number of elements processed by each vector instruction. Instructions producing vector shape results, write zero to destination vector elements past the end of `vl`.

Any motivation on that?
`vl` is supposed to be the number of elements that are going to be changed. The quoted statement implies counter-intuitive side effect going past `vl`.
Elements past `vl` can just be left untouched and used later with different `vl` value.

vbx-glemieux, 2018-10-18:
see https://github.com/riscv/riscv-v-spec/issues/13#issuecomment-431079152

colinschmidt, 2018-10-18:
Closing in favor of keeping discussion on this issue in one place #13

closed 2018-10-18 by colinschmidt