riscvarchive/riscv-v-spec #19: vconfig instruction

chenwei-tw, 2018-10-18:
> Could use li + csrw instructions to write vcfg, or have a vconfig instruction with an 8-bit vregcfg immediate field to enable setting a base configuration in a single instruction.

Don't vcfg need to configure `vemaxw` and `vtypeen`?

> vemaxw[5:3]

By the way, may I ask what [5:3] means?

colinschmidt, 2018-10-18:
The `vconfig` is the current proposal does not write `vemaxw` or `vtypeen` (they should both be set before `vconfig`).

The `[5:3]` means a 3 bit slice of `vemaxw`. The first sentence in this section is an explanation: https://github.com/riscv/riscv-v-spec/blob/24d41e2adbd1557fbca10a42a6a068b140a46987/v-spec.adoc#211-vemaxw53

rkruppe, 2018-10-18:
@colinschmidt 

> The vconfig is the current proposal does not write vemaxw or vtypeen (they should both be set before vconfig).

A bit tangential but now that you mention it... this makes sense given the limited immediate bits, but it means that in typical code one would still need two instructions for setting up the vector unit (`vconfig` + `csrw vemaxw`). Is the dedicated instruction even worth it then, compared to using just `li` + `csrw` (as was suggested before to expand the bits available for configuring the number of s/d/q registers)?

colinschmidt, 2018-10-18:
Its certainly worth thinking about how to make vconfig most useful, and we probably aren't quite there yet.

Its worth noting some implementations might most often be in a single `vemaxw` regime.
For example in RV32V keeping `vemaxw` at 8bit still enables you to configure the machine with registers up to `XLEN` with the single `vconfig` instruciton.

chenwei-tw, 2018-10-19:
@colinschmidt Thanks for your reply! And the v-spec is more clear now ;) 

May I ask two more questions?

1. Would the `vconfig` be a privileged instruction? If so, it may be wrapped a system call to use?

2. I am very curious about the scenario of such a number of configurations. If some libraries compiled for certain configuration and my program compiled for another,  does it means I need to call `vconfig` every function call?

chenwei-tw, 2018-10-19:
@colinschmidt 

> The [5:3] means a 3 bit slice of vemaxw. The first sentence in this section is an explanation

Could I understand that `[5:3]` is the position of `vemaxw` in the `vcfg` CSR?

chenwei-tw, 2018-10-19:
According to the `inst-table.adoc` and spec,

Why does `vconfig` need vd, vs1 and even mask ?

chenwei-tw, 2018-10-27:
Does `li + csrw` update the MAXVL and initialize VL CSR? Is `li + csrw` able to configure the different configuration out of `vfastcfg`?

colinschmidt, 2018-11-06:
Yes `li + csrw` sets `MAXVL` and initializes `VL` to `MAXVL`.
```
li a0, CONFIG
csrw vcfg, a0
```
The instruction sequence above writes to `vcfg` which contains the fields, `vregmax`, `vregd`, and `vregq`, `vsew` and `vtypeen`. These fields enable all configurations available in the base and therefore are a superset of the `vfastcfg `configurations.

closed 2018-11-06 by colinschmidt