riscvarchive/riscv-v-spec #21: huge vector lengths

brucehoult, 2018-10-18:
I propose that vsetvl with the hi bit of the length set should either trap, or else treat the length as zero.

If vemaxw is 8 then such a vector will take more than half the entire memory address space. If vemaxw is bigger than 8 bits then the implied vector will be larger than the memory address space.

Either way, it is almost certainly a programming error, and quite likely one caused by calculating what is effectively a negative vector length.

aswaterman, 2018-10-18:
Copying my response from the email thread:

This proposal seems problematic for auto-vectorization.  While it's unlikely to have a vector of 2^31 bytes in RV32, it's certainly not forbidden.  (Some Linux distributions give 32-bit user-space 3 GiB of virtual-address space, so arrays can exceed 2 GiB.)  Unless the compiler can prove that N is less than 2^31, it would need to insert extra code to deal with the possibility.

Adding the extra branch in the signed case is a small price to pay to sidestep this issue.  Bill's suggestion to add vsetvls is also reasonable, if we conclude that signed vector lengths are sufficiently common.

rkruppe, 2018-10-18:
It is routine for optimizing compilers to assume that no allocation is larger than half the address space ([some examples here](https://trust-in-soft.com/objects-larger-than-ptrdiff_max-bytes/)) as that would makes a mess of far more important things than a single test before a vectorized loop, so this shouldn't impact autovectorization.

colinschmidt, 2018-10-19:
Is the common case for signed vector lengths, that the loop index is a signed integer, thus giving the potential benefits that has w.r.t. the C spec?

rkruppe, 2018-10-19:
Not quite. The comparison done by `setvl` is the comparison between the induction variable and the trip count, and if that comparison is signed in the source program then we could indeed use a signed comparison for setting vl (if we can't prove that it doesn't matter). But thanks to the "usual arithmetic conversions", the signed induction variable will be promoted to unsigned if it's compared to an unsigned trip count of same or larger bit width. So the relatively common cases of `int < unsigned` and `int < size_t` are actually unsigned comparison, where the existing `setvl` works fine.

Still, a signed trip count isn't uncommon either.

kasanovic, 2018-12-01:
The current spec treats AVL as an unsigned value.  We don't make vl directly writeable to simplify specification (vl can only hold values within range).

With current PoR, I'd assume compiler would first analyze any loop index calculations to get a needed trip count .  So question is not exactly if loop control variables involved are signed, but whether compiler can statically determine that trip count is not negative (e.g., int I; int j=-32; for (I = -57; I <j; I++)).

Supporting signed AVL, would not be too onerous, adding condition if AVL<0, vl=0.  This is a small but non-zero  cost.   The only reason to support both signed and unsigned AVL is if trip counts could be greater than half the address space size.  If this is deemed illegal anyway by language lawyers, then it would appear we should only have the signed AVL variant, as compiler never needs to worry about case where loop trip count has high bit set.

aswaterman, 2018-12-01:
I’m not comfortable eliminating the unsigned variant, since you’d need to agree across all languages you might autovectorize that memory allocations can’t exceed the half the address space. And the code to emulate the signed variant using the unsigned variant is quite a bit simpler than the code to emulate the unsigned variant using the signed variant.

It’s sufficiently cheap to provide both variants that it might be justifiable to do so, in order to eliminate the (uncompressible) loop-head branch in the signed-comparison case.

kasanovic, 2018-12-01:
The constraint is actually narrower in that that allocations of byte arrays can't exceed half the address space; anything with halfwords or more is fine.

That said, it is fairly cheap to provide both (except for documentation/verification cost etc).

rkruppe, 2018-12-02:
@kasanovic 
> With current PoR, I'd assume compiler would first analyze any loop index calculations to get a needed trip count . So question is not exactly if loop control variables involved are signed, but [...]

Determining the trip count statically often isn't possible (because it's truly dynamic, or computed in another function).

In cases with "weird" loop bounds such as iterating from -57 on (even if the trip count is unknown), it's already more convenient for a vectorizer to create a new canonical induction variable that start from 0 and steps forward by 1 and use it for loop control. In that case the unsigned compare works fine even if the source loop involved signed/negative numbers (some bookkeeping is needed but it's not specific to signed/unsigned, it comes from handling the unknown trip count and weird induction variable).

A signed vsetvl is most useful for a loop vectorizer in cases such as `int n = ...; for (int i = 0; i < n; ++i)`, and there handling `n < 0` by not entering the loop in the first place is rather simple and not even necessarily slower. So I second @aswaterman's recommendation to prefer the unsigned variant to the signed one if we have to choose one.

kasanovic, 2019-09-06:
By changing vsetvli rd, x0, imm to mean don't change vl, we need to use li t0, ~0; vsetvli x0, t0, imm to get maximum vector length.

kasanovic, 2019-09-20:
Decided to stick with unsigned vector length only.

closed 2019-09-20 by kasanovic