riscvarchive/riscv-v-spec #22: Where is `vconfig` instruction encoding/details?

aka-sps, 2018-10-18:
What is `vconfig` instruction? Macros or separate instruction?

If based on `csrw` macros: Which of the registers is hardcoded for temporary use?
If separate RVV instruction: Where can I find a detailed description and encoding?

`vcfg` also contains set of fields (`vregcfg`, `vemaxw`, `vtypeen`, `vxrm` and `vxcm`).
Which ones are initialized by this instruction?

You can see the specification attempts to squeeze configuration to eight bits, while the `csrwi` instruction allows only five bits. I-format supports 12 bits.

Why eight bits, not five, not twelve?

colinschmidt, 2018-10-18:
The current proposal's encoding has 8 bits of immediate space. 
It also assumes we make `vconfig` an instruction that uses the normal vector immediate format.

colinschmidt, 2018-10-18:
In the current proposal only `vregcfg` is initialized by this instruction see the following for how the immediate bits are mapped to `vregcfg`. https://github.com/riscv/riscv-v-spec/blob/24d41e2adbd1557fbca10a42a6a068b140a46987/v-spec.adoc#311-vfastcfg-immediate

8 bits is the largest immediate that will allow all instructions currently in the proposal to be encoded. If we were to reduce the number of instructions in the proposal or take more than a major opcode we could use larger immediates.
It would be trivial to use a 5 bit immediate for `vconfig` but most people seem to be arguing for larger configuration spaces not smaller ones now-a-days.

colinschmidt, 2018-10-18:
`vconfig` is in the encoding table as of d6cd899

colinschmidt, 2018-10-19:
Closing this. Further discussion of what fields the fast configuration mechanism sets will be in #19.

closed 2018-10-19 by colinschmidt

aka-sps, 2018-10-19:
@colinschmidt
Agree. 
`vconfig` definition is the last stage of definition configuration itself.

Thanks