riscvarchive/riscv-v-spec #23: vsetvl with 0 argument behavior

go-sc, 2018-10-18:
Section 3.2
> The vl setting must be:
>  1.  greater than 0, if rs1 is greater than 0

Setting `vl` to 0 is meaningless and may lead to an infinite loop.
Probably, this instruction should raise an exception.

go-sc, 2018-10-18:
@rkruppe 
> a correct strip-mined loop has to subtract VL from the number of iterations left and exit the loop if no iterations are left

That's exactly an infinite loop when `vl` = 0.

> This can even be useful because it sometimes allows eliminating a "would the loop execute any iterations at all?" check before entering the loop

It's not obvious for me, could you provide an example?

rkruppe, 2018-10-18:
Here's a very simple example loop that just copies `a0` bytes from the memory pointed to by `a1` to the memory pointed to by `a2` (basically `memcpy` except I argument order), omitting configuration instructions because they're not relevant here:
```
loop:
setvl t0, a0
vlb v0, 0(a1)
vsb v0, 0(a2)
add a1, a1, t1     # src pointer += VL
add a2, a2, t1     # dest pointer += VL
sub a0, a0, t0     # (1)
bne t0, x0, label  # (2)
```

If intially `a0 == 0` then this doesn't access any memory, it just clobbers some registers, doesn't access any memory, instruction `(1)` places 0 - 0 = 0 in `a0`, and instruction `(2)` exits the loop.

aka-sps, 2018-10-18:
Why not:
~~~
loop:
    beq a0, x0, endloop
    setvl t0, a0
    vlb v0, 0(a1)
    vsb v0, 0(a2)
    sub a0, a0, t0
    add a1, a1, t0
    add a2, a2, t0
    j  loop
endloop:
~~~

rkruppe, 2018-10-18:
That works correctly but executes more instructions per loop iteration (two branches instead of one). Usually it's better to do the check once *before* the loop and keep another copy of the same check at the end of the loop body. But even then it's an extra instruction that is not necessary if VL=0 is allowed.

Besides, the meaning of VL=0 falls out *naturally* from the general semantics. Disallowing it would mean special casing in the spec, in software, in compilers, and (as far as I can tell, I'm not an expert) even in most hardware implementations. I've seen no convincing reason for doing so, everything's simpler if VL=0 is allowed and behaves normally.

aka-sps, 2018-10-18:
Ok, it was a naive `while do`.

What about:
~~~
    j 2f
1:
    csrw vl,a0
    vlb v0, 0(a1)
    csrr t0,vl
    add a1, a1, t0
    sub a0, a0, t0
    vsb v0, 0(a2)
    add a2, a2, t0
2:
    bne a0, x0, 1b
~~~

Two instruction is less than full loop with zero `vl`.

IMHO `vl==0` is strange and dangerous case. Both for write and read `vl` CSR.

aka-sps, 2018-10-18:
In general semantics access to empty register is undefined, except non-scalar register zeroing useless side effect.

Scalar destination register will be filled by undefined value if `vl` is zero. Try, for example, `vdiv  vd.s, vs1, vs2`, when `vl==0`.
Also reduction operations will have strange result (reduction `xor`, `or`, `and` with predefined neutral element `0` can kill any previously calculated result).

colinschmidt, 2018-10-18:
See https://groups.google.com/a/groups.riscv.org/forum/?utm_medium=email&utm_source=footer#!msg/isa-dev/_GxaDKqhciI/mkbvbpZvAwAJ
for why most stripmine loops shouldn't have the check for `vl==0`.

rkruppe, 2018-10-18:
> Two instruction is less than full loop with zero vl.

Yes, omitting the check before the loop runs more code *if the trip count is actually zero*. Often it's expected to be non-zero though, and if the typical trip count is very small the extra branch can have measurable overhead. I'm not saying skipping the loop if the trip count is zero has no value, it does, but omitting the check can *also* be useful. (Perhaps not for `memcpy`, though, that example was chosen for simplicity, anecdotally a large fraction of `memcpy` calls copy zero bytes).

> IMHO vl==0 is strange and dangerous case.

With all due respect, you've not backed this opinion up with any reasons that hold up to scrutiny, and disallowing VL=0 out has costs (as explained before).

> Scalar destination register will be filled by undefined value if vl is zero. Try, for example, vdiv vd.s, vs1, vs2, when vl==0.

Integer division in the scalar ISA (with M extension) has a defined result and no side effects or traps, presumably the vector variant would match that. Regardless of that detail, scalar-destination V instructions are indeed not affected by VL (likewise for non-V instructions operating on the general purpose registers), so yes, not every loop can rely solely on VL=0 to handle the case of not executing any iterations. But so what? The trick is not applicable to all loops and not useful for all loops, but it is useful in *some* cases.

>  Also reduction operations will have strange result (reduction xor, or, and with predefined neutral element 0 can kill any previously calculated result).

In stripmined loops, VL=0 only ever happens if the loop didn't execute at all, i.e. if the *total* iteration count is 0, so it wouldn't lose results from previous iterations. Additionally, if a loop adds/ors/etc. up *zero elements*, the neutral element is typically the desired result, so everything works out fine. (And if not, again, nobody's claiming that you can or should write *every* loop like this.)

rkruppe, 2018-10-18:
Agreed that this should be explicit in the spec.

> Likewise, in the first-element case, it's not clear to me whether the operation should be entirely skipped over (after all, vl=0, so the source vectors contain no data), or whether there is an implied vl=1.

For scalar operations on the first lane of the sources that broadcast the result, the proposal is that VL is ignored entirely. This is clear in supplemental material like https://github.com/riscv/riscv-v-spec/blob/cd277aa919dba917e603b985a6bc571b3bc1773b/listing.adoc though that's arguably non-normative.

aswaterman, 2018-10-19:
IMO there is a reasonable case to be made for leaving elements undisturbed, and there is a reasonable case to be made for zeroing elements.  Some other reasons that the V task group chose zeroing (besides register renaming):

ECC.  For long-vector machines in modern process technologies, where the vector regfile will be mapped to SRAM, not having ECC is not an option.  The overhead is too high to do ECC on byte granularity, though.  If you want to preserve unmasked elements rather than zeroing them, and the elements are narrower than the ECC granularity, then the partial register writes turn into read-modify-write sequences.  If you zero both the unmasked elements and VL .. MAXVL-1, this isn't an issue.

Also, the spec requires everything to be zeroed after a reconfiguration.  This is for determinism and for security (don't want other process' stale data from becoming visible).  Otherwise, OS would have to manually zero state on context switches.  Even if you chose that option, you'd need a fast way to zero the state.  Once you have that mechanism, you can extend it slightly to handle the zeroing of elements VL .. MAXVL-1, so that it doesn't impact performance on long-vector machines.  This does require a few bits of per-vector-register microarchitectural state.

rkruppe, 2018-10-26:
Aside: I have been commenting here under the assumption that vector-destination instructions zero the destination with VL=0, I learned today this isn't settled yet and might also be defined as a proper NOP that leaves the destination alone.

colinschmidt, 2018-10-29:
Based on the 10/26/2018 WG meeting the behavior when VL=0 will be consistent with leaving the elements undisturbed and therefore be a true NOP, except for instructions with the scalar-destination mask field.

closed 2018-11-01 by colinschmidt