riscvarchive/riscv-v-spec #24: separate floating-point instructions from Base

vbx-glemieux, 2018-10-18:
Like the integer instruction set, the base vector instruction set should be layered with optional extensions.

In particular, in embedded applications, integer computation is sufficient. Hence, I suggest all floating-point instructions (and conversion instructions) be placed into an optional vector floating-point extension.

This could go further, in that some vector integer calculations are not strictly necessary, such as divide or indexed load/store, but it becomes increasingly difficult to determine where to draw the line.

However, floating-point support should be made optional. The vector floating-point extension itself could possibly be divided up as well. For example, a smaller version that supports only half and single-precision, and a full version that offers all precisions.

aka-sps, 2018-10-18:
Additionally: There are no 16-bits floating point in the base F/D ISA, but 16-bits floating point is mandatory format in RVV.

vbx-glemieux, 2018-10-18:
> Additionally: There are no 16-bits floating point in the base F/D ISA, but 16-bits floating point is mandatory format in RVV.

it's not too difficult to restrict a single-precision unit to output 16b instead of 32b, so i don't have any problem with adding this functionality to a single-precision floating-point vector unit.

however, it takes a lot more area to implement double-precision arithmetic.  multipliers are O(n^2) and shifters are roughly O(n log n) where n = # of bits. plus these FPUs are replicated (one per lane), so it's a substantial area cost, hence my suggestion that the single-precision and below case be separated out for possible lightweight implementations.

aka-sps, 2018-10-18:
First of all: FPU 16-bit also should be available for flexible mapping of operations to CPU/vector unit, for verification, for debugging and testing.

Number of stages, rounding etc, in optimized `fdiv` and `fsqrt` can be significantly different in comparison with 32b.

Of course, we can use common summators, multipliers, etc, but control logic for subnormals, overflows and rounding should be different.

And last, but not least:
conversion from 16b to 32b, 32b calculations and reverse conversion to 16b save the memory/register space, but need more time and energy.

vbx-glemieux, 2018-10-18:
the scalar ISA is already fixed and doesn't include FP16. it could be added as an optional extension, and perhaps made mandatory if a core has the floating-point vector extensions.

but there is little advantage to adding FP16 to the scalar core. on the vector unit, FP16 allows an implementation the choice whether to double the number of parallel lanes when operating on 16b floats. this doesn't make sense with FP16 on the scalar core unless you create a bunch of subword SIMD instructions.

if there is significant overhead to provide both FP16 and FP32 for instructions like fsqrt, then an implementation can choose to implement fewer lanes for those particular instructions (thereby saving area at the expense of runtime). having the freedom to tailor the cost/performance for different implementations is a good thing.

colinschmidt, 2018-10-19:
This was written in the original latex document and we have migrated that into the ascii doc as of fa13752
@alonamid Will update the operation table soon.

vbx-glemieux, 2018-10-21:
Thanks @colinschmidt -- I had forgotten that text was written before. Good catch to notice that it needed to be added to the present version.

I think the specification as written here is logical but overly restrictive. For example, I can imagine a case where a scalar cpu that supports G isa is paired with a V-spec engine that supports only integers, or only integers and FP16, for NN applications. The full FPU in the scalar host allows certain arithmetic to be done efficiently without the slowdown or bloating of software floating-point libraries. Likewise, adding full single or double precision support in the vector engine would cause area bloat where it is not wanted.

To be clear, designers should be allowed to support some level of vector data types <= scalar data types (where <= is a subset operation). The spec currently forces support to be ==. (I don't think it makes sense to allow >, because the cost of adding support in the scalar host is small compared to adding it to the vector unit, and you might always want a "fall-back" capability on the scalar host, eg when verifying/debugging vector code).

rkruppe, 2018-10-21:
Unfortunately, as far as I know, this still isn't written down anywhere official let alone implemented, but the direction RISC-V has wanted to go for a while now is that the ISA specs list architectural state, individual instructions, etc. whereas relations such as "these instructions together form the $foo extension", "if these instructions [e.g., scalar FP] are present, those instructions [e.g., vector FP] must be present as well", etc. go into *profiles* which standardize common combinations of features without ruling out other combinations. It should absolutely be possible to implement a core which has RV32G and the integer part of V &ndash; though there may not be much of a pre-existing software ecosystem for such obscure combinations. The question at hand is whether the standardized profiles (that many people will compile for, or at least consider compiling for) should specifically include an option for "scalar FP but no vector FP" combination or whether it's something any vendor can do on their own.

vbx-glemieux, 2018-10-21:
This is what was added to the spec by https://github.com/riscv/riscv-v-spec/issues/24#issuecomment-431205223:

When the base V extension is
added to a base scalar ISA, it must support the vector data element
types and operations implied by the supported scalar types as defined by
tables <<velemtypes-int>> and <<velemtypes-fp>>.

[#velemtypes-int]
.Supported Fixed-Point Formats
|===
  |RV32I  |I8, U8, I16, U16, I32, U32
  |RV64I  |I8, U8, I16, U16, I32, U32, I64, U64
  |RV128I |I8, U8, I16, U16, I32, U32, I64, U64, I128, U128
|===

[#velemtypes-fp]
.Supported Floating-Point Formats
|===
|F      |F16, F32
|FD     |F16, F32, F64
|FDQ    |F16, F32, F64, F128
|===

So, any CPU that supports FD in the scalar core must also support FD in the vector core.

I would say this wording should be reversed:

When the base V extension is
added to a base scalar ISA, the scalar base ISA must support scalar operations for the vector data element types and operations implied by the supported vector types as defined by
tables <<velemtypes-int>> and <<velemtypes-fp>>.

colinschmidt, 2018-10-29:
This was clarified to be a profile issue in #42.
Considering this closed based on that.

closed 2018-10-29 by colinschmidt