riscvarchive/riscv-v-spec #26: Reversing operands for`vsubi`.

aka-sps, 2018-10-18:
Functionality of almost all cases of current variant of `vsub` can be replaced by `vaddi` with negated `imm` operand.

More useful case to use `vsub` with reversed order of operands:

~~~
vsub vd,imm,vs1
~~~

~~~
vd[i] = sext(imm) - vs1[i]
~~~

Negation operation in base ISA use `zero` register `sub rd, zero, rs2`.
To support negation in vector ISA need temporary filled zeros vector register or more than one instruction.

`vsubi vd,imm,vs1` form can be more useful, than original form.

aka-sps, 2018-10-19:
@vbx-glemieux 
`vsub-->vsubi` fixed
Thanks!

colinschmidt, 2018-10-23:
There is no `vsubi` in the current proposal. Adding another immediate instruction in the current encoding would eliminate the remaining free encoding space. A unary negation instruction would be more manageable in terms of encoding space but we should verify whether others think it is common enough for inclusion.

aka-sps, 2018-10-23:
@ colinschmidt
You can see `vsubi` [here](https://github.com/riscv/riscv-v-spec/blob/master/listing.adoc).

Please update and synchronize.

colinschmidt, 2018-10-29:
Added a unary negation operation (`vneg`) in #42 for the most common use of `vsubi`.
Closing this based on that PR.
Also reminding everyone that listings.adoc is not updated and does not reflect the current proposal.

closed 2018-10-29 by colinschmidt

vbx-glemieux, 2018-11-01:
Using vneg is a low-performance option for a potentially frequent operation.

We should add a vsubi instruction, albeit slightly different from how originally presented in this thread:

vsubi vd, imm, vs2
vd{i} = sext(imm) - vs2

The original proposal used vs1, which is wired to operand A of the ALU. Switching to vs2 does not require rewiring all of the vector ALUs. It may affect instruction encoding, particularly which bits are used to represent the immediate. For consistency, it is recommended that vaddi use the same encoding:

vaddi vd, imm, vs2
vd{i} = sext(imm) + vs2

The issue now is whether to add a vneg operation, which can be accomplished with vsubi vd, 0, vs2, or to add a vsubi operation. Unfortunately, vneg cannot be used as a direct substitute for vsubi, and instead must use a 2-instruction sequence of vneg followed by vaddi.

In tight inner loops, eliminating the extra vneg instruction has a big performance impact. The negative consequences of using vsubi instead of vneg are much more minor.

aswaterman, 2018-11-01:
I don't object to adding this instruction, but should point out that if
this is in a tight loop, you can register-allocate the immediate and use
vsub, rather than using vneg + vaddi.