riscvarchive/riscv-v-spec #27: Impossible to keep track of splat results

vbx-glemieux, 2018-10-18:
If you want to use a scalar src value (broadcast to all lanes) along with a vector src value, the current spec implies the result in the src vector (treated as a scalar splat) must be the result of a previous scalar shape operation.

Specifically, according to the table heading in section 5.2: "Dst Scalar/Vector Behavior: vector dst, scalar src (src was written previously as a scalar using 00 mask bits)"

The problem with this is that an implementation must now track which vector registers were the result of a scalar shape write, and that this "tracking" is lost if the register is ever spilled to memory and restored (eg, due to an interrupt, context switch, or whatever).

There is no easy fix to this -- if you want to allow scalar source values to be stored in element<0>, then you need an extra mode for vsrc1 and/or vsrc2.

colinschmidt, 2018-10-19:
The 5.2 table had a mistake. AFAICT the only place where scalar source made a difference in the base rows was in the store instructions. We have corrected this to also be a splat in 8113576.

Let me know if this is ok to close.

vbx-glemieux, 2018-10-19:
I don't think that's the issue I was relating to, but I agree with the correction.

The heading of the last column of the table says "(src was written previously as a scalar using 00 mask bits)". This implies the only way to use a scalar src is if the last write to that vector register was a scalar shape.

I think this is too restrictive. After spilling a splat result to memory and reloading it, it fails to retain this "scalar shape" because changing vector length does not extend the splat to maxvl. But, before the spill, the register would automatically be extended to any vl and could be used as a scalar operand that is broadcast to all elements.

Thus, a spill must know that, upon loading, the value needs to be re-splat.

colinschmidt, 2018-10-23:
If the spilling and restoring is done by a compiler then it can use the "scalar" load to restore the "scalar shape".
If the spilling is done by an OS (i.e. context swap) it would need to `vsetvl` to `MAXVL` in order to guarantee saving all of the state above `vl` in the vector registers (see https://github.com/riscv/riscv-v-spec/blob/204db8c48f2bde1f3c12ba79e8df0d7777b9538a/v-spec.adoc#32-vl-and-vsetvl-instruction)  and this procedure would also restore the "scalar shape".

colinschmidt, 2018-10-23:
A more precise but slightly more annoying link https://github.com/riscv/riscv-v-spec/blame/204db8c48f2bde1f3c12ba79e8df0d7777b9538a/v-spec.adoc#L332

vbx-glemieux, 2018-10-23:
Sorry but there are problems with your suggestions:

(a) the compiler doesn't know which registers have a "scalar shape". It does know the registers used in the local function, but not across function call boundaries (especially when they span compilation units). hence, the compiler cannot use the scalar load to restore the scalar shape.

(b) the OS does not know which registers have a scalar shape. even if it spills MAXVL elements, when it loads the data back it can't know which ones were a scalar shape.

The way the last column of Table 5.2 (now Appendix B vector-op-types.csv) is written, some instructions must be able to determine whether the source is a scalar or vector because they behave differently in each case. It is not sufficient to simply have the same value replicated in all elements. Hence, each register needs a state bit that indicates "scalar" or "vector" result, and right now that state information cannot be saved/restored by a context switch or the compiler (well in some cases the compiler knows to regenerate the information, but not all cases).

colinschmidt, 2018-10-23:
The idea behind the "scalar shapes" is that they should be indistinguishable from a vector splatted with the same value.
The last column of the table should say splat for all Base rows but could say something different for VTypes rows.

To correctly save and restore any vector register, that you have no knowledge of, you will need to set `vl` to `MAXVL` and then save the full "currently configured element width". This is not because of the scalar shapes but because of this:

> Changing the value of the vl register (either using vsetvl or using CSRRW) does not mutate the register file state. i.e. - values in the register file are preserved.

This is also the reason for the minor difference in writing "scalar shapes" @rkruppe noted.

All this enables a microarchitecture to, if they wish, track scalar shapes and minimize register file reads and writes, but by no means does it require this tracking. At least that is the intent.

colinschmidt, 2018-10-23:
Yes. That is a correct summary. Hopefully we can identify all the confusing parts of the current spec and amend them.

Thanks.

PR #57 (2018-11-06): Split vector-op-types.csv into a base and a vtypes table
There is a lot of overlap, but some fields only make sense with vtypes, so combining them can cause confusion. Additionally, the headers in the base table can be simplified a bit now.

Closes #27
commit 09269369 (2018-11-06): Split vector-op-types.csv into a base and a vtypes table

There is a lot of overlap, but some fields only make sense with
vtypes, so combining them can cause confusion. Additionally, the
headers in the base table can be simplified a bit now.

Closes #27
closed 2018-11-06 by colinschmidt