riscvarchive/riscv-v-spec #28: Byte offsets in `vlx`

aka-sps, 2018-10-19:
Vector of byte offsets calculation from vector of indices is extra operation, especially for naturally aligned load/store operations.

Unaligned data access is not common feature of RISC-V and not RISC-style at all.
Most of sizes for memory write operations are significantly system-dependent (for example, from system bus) and common solution is: split non-standard unaligned transaction into the set of standard size and alignment transactions.

Additional problem - page fault during cross-page boundary memory access (formally unresolved problem even with RVV precise-to-element trap support).

IMHO: base RVV spec should define only naturally aligned memory operations to prevent incompatible programming tricks. And base RVV ISA `vlx` instructions should operate with indices only instead byte offsets as in usual C pointers arithmetic.

Additional motivation:

~~~
double arr1[128];
double arr2[32768];
~~~

Indices for `arr1` are 8 bit, for `arr2` 16 bit, but byte offsets should be 16 and 3 bits respectively. Greater elements size - shorter vector register, less `MAXVL`, more iterations.

colinschmidt, 2018-10-23:
The indexed vector ops follow the strided vector ops.
Using byte indices enables the reuse of the index vector to gather multiple members of a struct throughout an array of structs.

```
struct x {
  uint64_t a;
  uint64_t b;
}

struct x array [vl*N]
size_t indices[] = {5, 10, 15, 7};
vsetvl a0, 4
vld vl, 0(indices)
vsli v1, v1, 4 #log2(sizeof(struct x))
vlx.w v2, 8(array), v1 #offsetof(struct x, b)
vlx.d v3, 0(array), v1 #offsetof(struct x, a)
```

aswaterman, 2018-10-23:
Byte offsets are also useful for gathering from an array of pointers. In
this case, the base is x0, and the offset is the pointer value.

aka-sps, 2018-10-23:
@aswaterman
Your argument is more significant. Thanks!

closed 2018-10-23 by colinschmidt