riscvarchive/riscv-v-spec #29: No vfmv instruction, unclear result when destination is scalar shape for vinsert and vfmv

vbx-glemieux, 2018-10-19:
Section 12.1 discusses vfmv.f.v and vfmv.v.f

These mnemonics are not in the new vector-op-types.csv table.

    vfmv.v.f  vd, rs1, vm   # vd = rs1
    vfmv.f.v  rd, vs1       # rd = vs1[0]

If vfmv.f.v is allowed, the behaviour written as "vd = rs1" is unclear because the LHS is a vector: are all elements splat with rs1? Is the result treated as a scalar shape? Or is only element 0 written with the value of rs1?

The integer counterpart, vinsert, specifies the destination element using rs2:
    vinsert vd, rs1, rs2, vm  # vd[rs2] = rs1

If vinsert is run as a scalar shape destination, I'd imagine the value in rs2 is ignored and a splat is performed while writing to vd<0>?

colinschmidt, 2018-10-23:
This instruction was not in the previous document, in the instruction listing, or discussed in the meeting notes, and was left from a previous version of the document.
We will remove until further discussion on the merit of saving the single `fmv.x.f`.

As to your second question yes it should be clarified/corrected that `vinsert` with scalar shape destination splats the x register into `vd`

See 98d1e5a for these changes

vbx-glemieux, 2018-10-23:
I think there may still be errors in the VINSERT row:

there should be 2 G registers (not 1)

it should not be illegal when dst is different width than src, because src is a GPR (width XLEN), and dest is a vector (width maxew).

Guy

colinschmidt, 2018-10-29:
`vinsert` table entry was fixed in #42 
`vfmv.v.f` is essentially `vinsert` from the scalar floating point registers which saves a single scalar instruction. We still need a wider discussion, likely in the next WG meeting.

commit d165905d (2018-11-01): Add vfinsert and vfextract for moving bits to/from fregs

Fixes #29
closed 2018-11-01 by colinschmidt