riscvarchive/riscv-v-spec #30: Notation [] used for both bit positions and element numbers

vbx-glemieux, 2018-10-19:
Within the spec, the square brackets are used with two different meanings.

Sometimes they imply bit positions, eg [XLEN-1 .. 0]

Sometimes, they imply a vector element number, eg vd[i] = vsrc1[i]

I'd recommend a distinct notation be used for the two cases. Personally, I have been using <i> to denote element number.

aka-sps, 2018-10-19:
What about SytemVerilog syntax?
`<` and `>` are comparison terms.

In c++ there are problems to parse template arguments in `<  >`  which contains `>`.

Usually needs additional `( )` around expression with `>` to avoid erroneous temlate parameters list termination.

Verilog have own inconveniences. For example: `<=` can be assignment and can be "less or equal" 

C operation `[ ]` is a element indexing.
May be `{k..m}` notation as bit range will be better.

aswaterman, 2018-10-19:
The entire RISC-V spec uses [ ] for bit ranges, so we should be consistent here.  Using something else for element indexing makes sense.

colinschmidt, 2018-10-23:
Since we are using `< >` for things like `vtype<n>` where it acts somewhat as a template, `{ }` seems reasonable for element indexing.
Eventually this nit will be fixed...

vbx-glemieux, 2018-10-23:
At the beginning of section 5 on masking, the example comment uses notation v0.LSB to extract the least significant bit. This is incongruous with the use of [0] elsewhere in the doc for bit positions.

I'm ok with v0{i} for element i. we could also use a dot notation, like v0.i, but then the above example of .LSB would need to be cleaned up.

aswaterman, 2018-10-23:
Yep, dot notation seems fine too; either way we should use [0] rather than .LSB.

go-sc, 2018-10-24:
> #### 3.1.1. vemaxw[5:3]

This is not a correct notation since `vemaxw` has only 3 bits.

It is also a little bit confusing to describe register's length in terms of ranges because it is not clear what is being specified: the whole register or just its fragment. Probably the length should be mentioned explicitly without slice notation like in 3.1.2:

> #### 3.1.2. vregcfg
> vregcfg is a 12-bit field...

colinschmidt, 2018-10-29:
This was closed in #39.
Please submit a separate issue for the `vemaxw[5:3]` comment @go-sc  if its still an issue.

closed 2018-10-29 by colinschmidt