riscvarchive/riscv-v-spec #31: Forcing clear of all registers upon vconfig considered harmful to performance

vbx-glemieux, 2018-10-19:
The spec currently requires ALL vector registers to be cleared upon vconfig. The rationale is for security on a context switch, and determinism.

This is rather heavyweight, and forces all applications to spill vector registers if they simply want to change the max vector element width -- that is, if they wish to change the amount of available parallelism between stages of their application, all vector state must be spilled to memory.

I don't understand how this helps determinism, so I won't comment upon that.

However, to preserve the security requirement while also allowing a lightweight implementation, I suggest we have two different vconfig instructions. One, which will be used upon a context switch, will do the required clearing of all vector registers. The other, which will be used by compilers and programmers, will preserve vector data in-place. Sure, it may be hard to access that data (since it won't be aligned to the new max element widths), but this only renders is temporarily useless until a restoring vconfig reverts it to the prior state. More importantly, this will save unnecessary vector spilling.

This would also allow some relief to compilers. In an email to vector-ext with the subject "RISC-V vector extension cauldron discussion", it was suggested that "an abi must consider how vconfig is to be managed across function boundaries and with separate compilation". In particular, there was a suggestion that "the callee must be able to reset to a previous vconfig", and "the callee should be able to call-save registers". Forcing a clear of all registers forces spilling all registers to memory. If they are left in place, then a callee only has to spill those registers it needs to use, which could be a dramatic subset.

rkruppe, 2018-10-19:
The V extension is designed to allow implementations to reorganize their physical register file in response to configuration changes to offer a greater maximum vector length for some configurations. In this context, the choice is not between clearing all registers or leaving them alone, it's between clearing all registers or making each register contain an unpredictable scramble of all the bits previously stored in the register file (how it's scrambled falls out from how the register file is reorganized). This is the dichotomy the "security and determinism" rationale refers to.

Even ignoring the details of how the bits in the register file are (re-)organized, the reason we have the configuration in the first place is to vary the maximum vector length depending on the configuration (i.e., the needs of the application). The issue text says as much in this part:

> [applications] simply want to change the max vector element width -- that is, if they wish to change the amount of available parallelism between stages of their application

This property does not seem compatible with wanting the contents of some registers to be preserved: Since all vector registers have the same number of *elements*, changing the element width of a register changes the number of bits required to store that register. The only way to make that work without having to eat into other registers' storage is if each vector register *always* pessimistically has ELEN bits per vector element reserved, even if a smaller element width is configured. But in that case the implementation can't offer increased longer vectors to code working with smaller element widths, and there wouldn't be any reason to bother with the dynamic configuration in the first place.

There *is* a way to make "operate on some registers with different data type sizes with different amounts of parallelism without disturbing unused registers" work, but it means making the vector register size measured in bits rather than elements and not varying this bit size. This is the route Arm has chosen with SVE, and this decision has many far-reaching consequences for all aspects of the ISA, how it is programmed, and how compilers generate code for it. These consequences are not at all mentioned in the issue text, so I'm really unsure whether this is really the route you have in mind. Please clarify.

vbx-glemieux, 2018-10-21:
There are two goals of vconfig that are at odds with one another:
-- use smaller element sizes, for greater parallelism
-- use longer vector lengths, for more efficient vectorization

In my experience, when possible, using smaller element sizes produces greater speedups, and should be the first priority for acceleration. Hence, it should be treated as a first-class priority by the ISA. In cases where this is not possible, then using longer vector lengths should be next.

By conflating these two issues into a single vconfig, it causes the problems noted above by @rkruppe.

There is no reason that changing the element size on the fly would cause an "unpredictable scramble of all the bits previously stored in the register file". The vconfig spec currently allows double and quad-word organizations, and it's very clear from the vector register numbering strategy that it is intended to join adjacent vector register numbers together to support the d and q configurations. From this, it is easy to specify "how the data will be reorganized" after a vconfig that toggles between support for just s and adding d and q registers. It is also easy to extend this specification on how the data will be reorganized when the element width is changed.
 
The primary use-case that I'm trying to solve is computing on smaller element widths for as long as possible (increased parallelism), and then periodically shifting over to wider computation (eg, to accumulate into a wider element to avoid overflow) and back to the smaller element width.

In the vector code that I've written, it is quite common to do this mixed-width computation. The V-spec as is currently written requires a vconfig to the *widest* used element width, and the use of narrower vlb/vlh instructions to load narrow data into wider vector elements. This does not give the parallelism advantage that would otherwise be possible.

Here is an example computation for edge detection:
(a) deinterleaved 8b pixels packed into rgba format (8b computation)
(b) extend 8b pixels into 16b while performing rgb-to-luma conversion (mixed 8b/16b source, 16b destination). rgb-to-luma requires fixed-point multiplication by constants (luma = a*r + b*g + c*b).
(c) perform 3x3 filter on source image for horizontal-edges; this can be done using 16b source for accuracy, or source can be converted to 8b for speed)
(d) perform 3x3 filter again for vertical edges (same as (c) above)
(e) compute magnitude from h-edges and v-edges, ie mag[i,j] = x[i,j]*x[i,j] + y[i,j]*y[i,j]. again, source data can be 8b or 16b for speed or accuracy
(f) convert magnitude back to rgba packed pixel for display (src is either 8b or 16b, output is 8b-packed)

As you can see, most operations would be 8b, but in places 16b can be used when precision is required (rgb2luma). The V-spec discourages this type of mixed-width computation, and makes it all run at 16b slowness.

rkruppe, 2018-10-23:
Oh, you're talking *only* about configuration changes that fuse or split registers, not changes to vemaxw or vregmax? Then limiting the clobbering to the registers being fused/split is more reasonable, though it still implies extra restrictions on implementations. It makes what is currently an implementation option mandatory, ruling out simple microarchitectures that don't perform such fusing (or only perform it in some cases) as well as those that would want to reorganize their register file in different ways. I'm not the best person to judge how big of a burden this is, but to me it seems significant.

On the flip side, how big are the benefits? The proposal with d/q registers readily supports mixed-precision computation, it just requires setting vemaxw to the smallest width you use and reserving some d/q registers up-front. If the implementation supports it, that will give the increased parallelism you want (confusingly embodied by a larger VL). Admittedly, reserving the d/q registers up-front does reduce the registers available for code that only needs s registers -- though d/q registers can be used for narrower data too, so each d register added effectively costs one s register, not two.

With respect to the compiler/ABI problems you cited earlier: being able to fuse and split registers without affecting other registers would *not* solve that, because one would still have to fix both vemaxw and vregmax to be able to do that. Any choice of those settings will be suboptimal for many functions, so such an ABI would still be mostly useful for vector function calls (though admittedly defining the vectorcall ABI becomes a bit nicer if we don't need to worry about how many d/q registers to reserve).

For the normal C ABI, the incentives to allow the caller to change vemaxw and vregmax are unchanged -- and altering those still wipes all vector unit state if I understood you correctly, so the problem of having to save all-or-nothing still exists.

kasanovic, 2018-12-01:
New spec exposes all state bits so has eliminated this concern.

closed 2018-12-01 by kasanovic