riscvarchive/riscv-v-spec #35: Ambiguity in the spec: Is "vl" a read-only or read-write CSR?

vbx-glemieux, 2018-10-21:
From the spec, there is an ambiguity to implementers whether the vl CSR can be written with CSR instructions:

3.2. vl and vsetvl instruction

Regular CSR instructions can be used to read and write vl.

Note | It is sufficient to make vl a read-only register, since the vsetvl instruction can be used to re-populate the register during context swaps.

a0u, 2018-10-23:
The note was added as a reminder to discuss whether `vl` should be made read-only as a minor simplification.  The original LaTeX document specified `vl` to be WARL, but that decision predated having a dedicated `vsetvl` instruction.

vbx-glemieux, 2018-10-23:
Thanks for the history @a0u.

This issue should remain open until fully resolved (ie, the discussion needs to happen, a decision needs to be made, and the ambiguity removed from the spec).

a0u, 2018-10-24:
ddb35cfa62bfb7ef28b2c633922186025909713f refines the `vl` setting rules in accordance with @rkruppe's observation.

Additionally, I have slightly relaxed the monotonicity requirement to be piecewise-defined: over `[0, MAXVL]` (trivially satisfied by `vl = AVL` for `AVL <= MAXVL`) and separately over `(MAXVL, +infinity)`.  This permits setting `vl = ceil(AVL / 2)` for `MAXVL < AVL < 2*MAXVL` as described in the original LaTeX document.

a0u, 2018-11-01:
54f3051ef52c5298ac164dc6011ecf7e17e5a266 updates `vl` to be a read-only CSR, now that `vsetvl` should behave identically to WARL semantics as originally intended  when `AVL <= MAXVL`.

closed 2018-11-01 by a0u