riscvarchive/riscv-v-spec #36: unclear on how vemaxw interacts with s/d/q registers (vregd, vregq)

vbx-glemieux, 2018-10-21:
I don't understand how vemaxw interacts with vregd and vregq specifications.

Using vregd and vregq, some vectors can be made twice as wide (double or quad words). The base registers are words.

If I change vemaxw to be 16b, does it mean the s registers remain 32b in width, but all data operations are 16b in width (leaving upper 16 bits of each element untouched)?  Or does mean the s registers are now 16b in size, stored in a packed format, and the allowed vector length is twice as long? I believe it means the latter, but this is not clear to me in the spec.

How, then, should the 'd' and 'q' registers be treated? Do they remain 64b/128b in size, but all data operations remain at 16b (so upper 48 or 112 bits of each element are untouched)? Or are they now 32b and 64b in size, but also twice as long? Again, I'd assume it would have to mirror the s case, so they should now be 32b and 64b in size but twice as long. 

Is it valid to have operations between differently sized registers (s, d, q)? It must be, since masking must always be done with v0 and that should be combined with a d or q register (making all the registers the same length means masks from v0 apply to d or q registers on an element-by-element basis). IF different sized register operations can be mixed, this implies there is some size conversion being done (when sources are mismatched, or destination mismatches) ==> this implies significant data shuffling between lanes is required to match the element locations of the larger size vectors. Also, should I assume this conversion is done in the same way as vcvt when downsizing, and when upsizing should sign or zero extension is being done (eg, on a vadd)?

chenwei-tw, 2018-10-27:
What happend if source operands are double-word registers but destination register is single-word register?

vbx-glemieux, 2018-10-27:
@chenwei-tw It is different for every instruction -- see https://github.com/riscv/riscv-v-spec/blob/master/vector-op-types.csv under column heading "Dst Type (dst_width = src_width / 2)"

chenwei-tw, 2018-10-27:
@vbx-glemieux Thank you. I got it.

chenwei-tw, 2018-10-27:
Considering the `vsew` = 32bit and quad-word vector register,  does it mean RV32V need to support at least one 128-bit ALU?

chenwei-tw, 2018-10-27:
> > Also, should I assume this conversion is done in the same way as vcvt when downsizing, and when upsizing should sign or zero extension is being done (eg, on a vadd)?
> 
> Integers are zero-extended or sign-extended based on the operation. Floating-point values become NaN-boxed when upsized.

According to the current spec, `vadd` doesn't offer {signed / unsigned} information.

chenwei-tw, 2018-10-28:
> It is different for every instruction -- see https://github.com/riscv/riscv-v-spec/blob/master/vector-op-types.csv under column heading "Dst Type (dst_width = src_width / 2)"

@vbx-glemieux Thanks for your response. But according to the current spec,

> Source operands can be of different widths, but with a difference of no more than a single factor of 2

there may be two different src_width. How to account for this situation? Such as,
```
vadd v1 (s), v2 (s), v3 (d)
```
Does v2 extend (*) to double-word and plus v3 then trigger ILLEGAL (Because dst_width = src_width / 2)?
Or v3 will be truncated to single-word and plus v2 then normally executed (Because dst_width = src_width)?

* However, I don't know whether is available to set zero-extend or signed-extend on each instruction.

a0u, 2018-10-30:
67e83b4e9788be8d6bad38d12d052190897e15b8 hopefully clarifies what the behavior should be when the widths of source/destination registers differ from the (non-polymorphic) operation's precision.

a0u, 2018-11-01:
478919e3fccf71b55fa44bbf62140491c46307ce revises the previous commit to clarify what "operation width" means for integer instructions, which are "semi-polymorphic".

I am closing this since the original issue was already addressed.

closed 2018-11-01 by a0u