riscvarchive/riscv-v-spec #38: missed performance opportunity: vredsum, vredmin, etc are all single-operand instructions

vbx-glemieux, 2018-10-21:
the reductions are all specified as single operand instructions, eg:

vredsum vd.s, vs1

there is a missed opportunity here to improve performance... many reductions are combined with another ALU operations.

in particular, summation should be mixed with multiplication:

vdotprod vd.s, vs1, vs2

should compute:  vd.s = sum_over_i( vs1[i] * vs2[i] )

otherwise, separate instructions are required to multiply and reduce, and this slows things down.

i don't think all forms of operations should be paired, but dot product is natural one to include.

aka-sps, 2018-10-21:
@vbx-glemieux 
IMHO "reduction" is only special "folding" mode for binary operations.

See: [FoldList](https://reference.wolfram.com/language/ref/FoldList.html) and [Fold](https://reference.wolfram.com/language/ref/Fold.html).

"Reduction" in terms of current spec is only  `Fold` with some predefined initial value for limited set of binary operations (`add`, `min`, etc.).

Proposed in previous version specification set of "reduction" operations is extremely limited and incomplete due to implicit "neutral element".

If we define reduction as "folding" with explicit scalar initial value we can implement both [std::accumulate](http://www.cplusplus.com/reference/numeric/accumulate/) and [std::partial_sum](http://www.cplusplus.com/reference/numeric/partial_sum/) for any kind of binary operation.

`Iota` ([std::iota](http://www.cplusplus.com/reference/numeric/iota/)) instruction can be simulated in similar manner.

Of course, `madd`, `msub`, `mnadd` should be folded only for `add`/`sub` (with last `vs3` operand), but `mul`  is ordinary `zip` operation (as in usual vector-vector elementwise `vmul`).

See: [std::inner_product](http://www.cplusplus.com/reference/numeric/inner_product/).

So dot product (`vd.s`) of two double precision vector (`vs1`, `vs2` of ) is similar to:

~~~
    vxor  vd.s, vd,vd  // init result (scalar ) register

1:
    // load source vectors parts
     ...
    // calculate partial result and add to previous sum 
    vfmadd.d<fold_mode>    vd.s, vs1, vs2, vd.s
   ...
   bne    zero, ..., 1b
~~~

Naturally, for the complete basis, the masking should be the orthogonal option for zip/reduction:
in many cases it is necessary to perform reduction only for masked/unmasked elements.
Note: `0` is not "neutral element" for signed integer `vredmin`/`vredmax`.

Floating-point `precise` (strong folding order) and` fast` (relaxed, ООО) modes for reduction should be available.

kasanovic, 2019-02-22:
Current spec 0.7 has initial scalar value input.

closed 2019-02-22 by kasanovic