riscvarchive/riscv-v-spec #43: Extended encoding for vew

grnewell, 2018-10-26:
For crypto we will need additional [very wide] element width encodings.  For example, for SHA-512 we desire an encoding that allows a "double" to achieve 1024 bits simultaneously with "single" elements that are 512 bits.

I believe there was an encoding proposed that included many powers-of-two and also those times 1.5.  The AES standard requires the key size 192 (in addition to 128 and 256); ECC uses key sizes like 192 and 384 (in addition to 256 and 512, with 384 being the minimum key size approved in the CNSA standard, therefore we expect it to be a popular choice).  RSA and DH use all powers-of-two and those times 1.5 with common usage from 1024 up to 16384, with 3072 being the minimum in CNSA. 

Also, there are applications such as machine learning that will benefit from narrow widths.  Again, there was a proposed encoding that covered these.

It makes sense to include the extended encoding table in this spec.

kasanovic, 2018-12-01:
New spec has up to 1024 bits in powers of two as standard.
Placeholder for extension vxsew[5:0] in vtype has been added at end of spec.

kasanovic, 2019-02-22:
Current spec supports up to SEW=1024 bits.  vtype register can be extended for larger types.

closed 2019-02-22 by kasanovic