riscvarchive/riscv-v-spec #44: Extended encoding for register mappings

grnewell, 2018-10-26:
The vfastcfg encodings for register mappings in Appendix A are a great improvement over the previously proposed mappings, and I understand the desire to keep them to 7 bits for the vfastcfg instruction so they fit in the immediate operand.  However, they are still restrictive for some applications.  Therefore, I would like to see an extended encoding that covers the entire reasonable universe of singles, doubles, and quads that can be loaded into an [extended] CSR, i.e., the number of doubles up to the maximum of 15 (4 bits instead of 3 as in the shorter immediate encoding), and quads up to the maximum of 7 (3 bits instead of 2 as in the shorter immediate encoding).

vbx-glemieux, 2018-10-26:
Not sure what you really gain here except save some code size in a preamble section of code.

Performance-wise, the register file is going to be cleared, so all data has to be loaded in from scratch anyways; if you want to preserve any data from before, this also has to be saved beforehand. You will almost certainly not be doing any vconfig inside an inner loop unless you deliberately want to flush all data from the register file quickly.

grnewell, 2018-10-26:
Note that the current more-compressed encoding for use as an immediate uses from 2 to 5 bits for the register "folding factor," and this should probably be retained for compatibility between the more-compressed and the extended encoding, even though a slightly more compact encoding could work.  Therefore, the proposed size of the extended CSR would be 5 + 4 + 3 = 12 (encoded folding factor + #doubles + #quads).

grnewell, 2018-10-26:
@guy:  There is a contingent, as reflected in the current version of the spec, that wants to be able to do a fast configuration with one instruction using an immediate.  I agree with you that this is not a great time-saver, but it seems to be the sense of the group to have this [vfastcfg]  instruction.  With several instructions, one would have access to the extended encoding I propose.   If we dumped the vfastcfg instruction and therefore attempts to make compatibility between it and the extended version we could conceivably save a bit or two in the extended version.

vbx-glemieux, 2018-10-26:
The advantage of vfastcfg, in my opinion, is that I can get a valid configuration of the register file without too much thinking. So I don't mind if the purpose of vfastcfg bits are changed a bit, as long as it lays out several possible fixed solutions that I can figure out quickly by looking at a table.

The whole process of producing a valid configuration is quite mind-numbing -- sure compilers can do it efficiently, but as a low-level programmer I hate this kind of stuff.

grnewell, 2018-10-27:
I have taken a stab at re-encoding ALL the possible legal S,D,Q element-size registers that follow the triangle pattern with doubles and quads aligned as expected.  The current plan-of-record encodes 76 register patterns in 7 bits.  My new proposal encodes an additional 40 patterns for a total of 136 in 8 bits using an encoding style very similar to the current 7-bit version.  If vfastcfg can hold eight bits for this purpose, then it seems for the price of one extra bit (currently reserved for future use) we can configure every reasonable pattern using an immediate operand, supporting the goal of a fast configuration without compromise or the need for an extended encoding requiring extra instructions to use.

See the attached spreadsheet:
[vregcfg-Microsemi proposal2.xlsx](https://github.com/riscv/riscv-v-spec/files/2521331/vregcfg-Microsemi.proposal2.xlsx)
Yellow highlights the current plan-of-record.  Green highlights the added patterns.  Lavender can be ignored:  it expounds all the illegal encodings under each of the systems shown (7-bit, 8-bit, and 12-bit).

grnewell, 2018-10-29:
It seems this and issue #8 are on the same topic

a0u, 2018-11-02:
If using all 8 bits, instead of splitting the `vregmax=32` configs across two prefixes (`11` and `101`) as in the spreadsheet, they can be consolidated under one prefix.  The other prefixes follow the existing 7-bit encoding with the MSB being 0.
```
1 xxx xxxx // vregmax=32
011 xx xxx // reserved
010 xx xxx // vregmax=16
0011 x xxx // vregmax=10
0010 x xxx // vregmax=8
00011  xxx // vregmax=6
00010  xxx // vregmax=5
000011  xx // vregmax=4
000010  xx // vregmax=3
000001  xx // vregmax=2
000000  xx // vregmax=1
```

grnewell, 2018-11-02:
@a0u 
Yes, this seems cleaner... thanks!

However, after today's discussion at the meeting, the 8-bit and perhaps even the 7-bit encodings may go away in favor of an even shorter one (for use in fastconfig) so more immediate bits are available for other purposes.  The discussion seemed to favor the non-variable-length 12-bit encoding for the extended form in the CSR register, even though it adds no more choices than the 8-bit variable-length encoding, just because it is more uniform and easier to understand and perhaps slightly faster to decode.  We wait on the Berkeley teams next proposal.

kasanovic, 2018-12-01:
Thread made moot by redesign in v0.6.

closed 2018-12-01 by kasanovic