riscvarchive/riscv-v-spec #46: Long-word addition

grnewell, 2018-10-30:
In March we discussed ways of doing long-word addition efficiently:  Here is the latest proposal from then:

	vadd3         vdest, vsrc1, vsrc2, vsrc3, vm 		# 3-input add, producing sum
	vadd3cout  vcout, vsrc1, vsrc2, vsrc3, vm         	# 3-input add, producing carry

I would like to resurrect this discussion again, for inclusion in the base Vector Extension.

colinschmidt, 2018-11-01:
We have no additional space in the 32 bit encoding for 3 input operations.

brucehoult, 2018-12-02:
Note that fadd3 is also desirable in scalar code as a counterpart to fmadd for use in things such as argument reduction for transcendental functions.

e.g.

fmul a,c,d
fmsub b,c,d,a

.. gives all the bits of c*d in a and b

fadd a,c,d
fneg t,a
fadd3 b,c,d,t

... gives all the bits of c+b in a and b.

Both of these are important operations that are very lengthy to do otherwise. IEEE 754-2008 mandates fmadd and friends, but fadd3 is the poor cousin.

In the absence of fadd3 you need something like this (Knuth):

{double, double} TwoSum( double a, double b )
{
     x = a + b;
     q = x - a ;
     r = x - q;
     s = b - q;
     t = a - r;
     y = s + t;
     return { x, y };
}

aswaterman, 2018-12-02:
It is a little off-topic, but on the subject of fadd3: the next IEEE 754 standard will include the augmentedAddition operation, which is basically defined to return (head, tail) where

    head = roundTiesToZero(x + y)
    tail = (x + y) - head

Unfortunately, fadd3 is not of much help here!  In addition to needing the new rounding mode to compute 'head' correctly, fadd3 won't produce 'tail' correctly with respect to signed zeros and exceptions.

If I were to implement these operations while being conscious of opcode space, I'd (a) add the new rounding mode to fadd to compute head, and (b) add a new two-input instruction that computes tail from only x and y.

The story is analogous for the new augmentedMultiplication operation.

vbx-glemieux, 2018-12-02:
Why can’t you just use a sequence of widening additions, ie

# SEW = 8b
# 24b accumulate from 8b source
# 8b source in v5
# 24b dest, lowest 8b in v6, highest 16b in v8/v9
vaddw.wv v6, v6, v5 # 16b = 16b + 8b
vaddw.wv v8, v8, v7 # migrate MSBs to v8/v9
vxor.vv v7, v7, v7 # clear v7

to add wide sources, you would handle carry outs similarly without a
ternary add:

# SEW = 8b
# 24b add, producing 32b result, done bytewise
# 24b source in v2/v3/v4 and v6/v7/v8
# 32b dest in v10 (8b LSB), v12 (8b), and v14/v15 (16b MSB)
vaddw.vv v10, v2, v6 # 16b = 8b + 8b
vaddw.vv v12, v3, v7 # 16b = 8b + 8b
vadd.wv v12, v12, v11 # accum LSB carries
vaddw.vv v14, v4, v8 # add MSBs
vaddw.wv v14, v14, v13 # accum mid carries

using 32b mode might be seem to be more efficient, but since you do 4 times
as many byte ops per cycle it might just be a wash.

Guy

kasanovic, 2019-02-22:
vspec 0.7 has vadc/vsbc with carry-in/out implicitly in v0 mask register (operation is unmasked).

closed 2019-02-22 by kasanovic