riscvarchive/riscv-v-spec #49: Minor: Suggestion to unify vector instruction mode designation in assembler

aka-sps, 2018-10-30:
In current specification there are 3 kinds of vector instruction mode designation.

Masking modes are designated as additional literal operands, unmasked mode has no additional operand and scalar mode uses special suffix `.s` of destination register.

_Proposal:_ For the syntax unification let designate scalar mode similar to masking mode (similar to optional "RM" field in the floating point instructions of "F"/"D" extension, similar to "AQ"/"RL" ordering fields of "A" extension instructions, etc.).

We can explicitly define mode labels (`a` - all, `t` - for true mask, `f` - for false mask, `s` - scalar) for the last optional `mode` operand with default value `a`.

~~~~
    vadd   vd,vs1, vs2        # implicit "a" mode - all elements in range 0..vl-1
    vadd   vd,vs1, vs2, a     # explicit "a" mode
    vadd   vd,vs1, vs2, t     # masking, for elements in range 0..vl-1 where LSB of mask element is 1
    vadd   vd,vs1, vs2, f     # masking, for elements in range 0..vl-1 where LSB of mask element is 0
    vadd   vd,vs1, vs2, s     # scalar: broadcast first `vs1`, `vs2` elements result to all `vd` elements in range 0..MAXVL-1
~~~~

It can help to use more regular patterns of extended inline gcc assembler.

kasanovic, 2018-12-01:
It is planned to later add a 64-bit vector instruction set extension that would have a complete explicit predicate register specifier.  So, ultimately, there would be:
`vadd.vv v5, v5, v7, v9.t # Execute where v9[I].LSB=1`

Also, now
`vadd.s v5, v6, v7`
only read/writes element 0 of all involved registers - there is no implicit "splat".

kasanovic, 2019-02-22:
superseded by new spec.

closed 2019-02-22 by kasanovic