riscvarchive/riscv-v-spec #53: Remove vsew, vregd, vregq from Base, move it to VTypes

vbx-glemieux, 2018-11-02:
My proposal is around simplifying the Base specification by getting rid of the complexity surrounding operand sizes. It is not only simpler, but more flexible and orthogonal for compilers and software writers. It is also simplified by removing support for very large operand sizes.

This proposal solves most issues with #52. I believe would enable what Ken Dockser has proposed on the closed vector mailing list under the subject "Packed SIMD as a Vector base option". I believe he is also proposing more extensive changes, which an be raised as other issues. He has said this can be kept alongside the current vector spec use of vconfig, but I am suggesting abandoning that.

Changes:
1) remove vsew, vregd, vregq entirely; move them to VTypes
2) possibly redefine how vregmax works
3) add operand sizes the instruction encoding for eachh operand

Rationale:
1) vregd and vregq imply operations on 128b and 256b word sizes. it is not clear these huge word sizes are necessary in Base. They will be rarely used, but adding support for add/sub/mul/div at these sizes increases complexity of all implementations.

2) vsew adds complexity with hidden state for compilers, and makes it difficult to build  vector registers with different element sizes. It also makes it difficult to know the data layout in the vector data store (read below), which increases the complexity of all implementations and adds obscurity to software.

3) The complexity of understanding all of the legal modes around vconfig, and how vconfig changes how the register file must be interpreted, are entirely removed. Instead, what remains is a very simple basic set of operations that can be orthogonally combined (all combinations are valid), without the need to check for legality.

Details:

A much cleaner way is to make explicit the data layout options and restrictions on the hardware, and to encode operand sizes for each instruction as part of the ISA encoding rather than as a fixed mode.

Here's how such an idea would work. Each instruction operand uses a 2b encoding to represent the operand size: 00=8b, 01=16b, 10=32b, 11=64b. This drops support for 128b and 256b operands.

The ISA can allow a maximum of, say, 16 named registers V0 to V15. Why 16 and not 32? Well, it depends on the number of bits available in the instruction encoding, so a drop to 16 may be necessary. Personally, I have seldom seen the need for 32 named registers (not saying it's not important, just that I don't think I've used that many in my vector programming), and this saves 3b in the entire instruction encoding (1b each for dst, src1, src2) which will be used up by the 2b-per-operand encoding. Thus, the net cost is 3b.

Vregmax currently defines the number of named vector registers. To simplify, I would suggest allowing only divisions of 2 in the number of registers. Thus, if the spec says there must be 16 named vector registers, then vregmax can choose between 16, 8, 4, 2 (or 1) register, thus allowing longer maxvl. The same naming scheme as present for double/quad element sizes would be used, where dropping to 8 registers uses register names v0, v2, v4, ... v14 and dropping to 4 registers uses v0, v4, ... v12. Whether the spec allows or blocks (illegal instruction exception) access to the other register names at a smaller maxvl (to access half of a vector) remains to be decided.

Data layout is specifically defined by the new spec, and it is quite simple. Each named vector register always points into a vector data store at a fixed address (ie, each name corresponds to an address). The operand data type determines whether data read from the data store will be interpreted as 8b, 16b, etc. Data is stored in a packed format (always). So, it is valid to write 8b data to a vector register (using, say VL=8), and then read it out later as 32b data (say, VL=2); this is actually quite common when an application wants to extract bytes from an interleaved structure (eg, rgba pixel data, and you want to extract the g bytes). Using a logical packed data format is very straight-forward.

This keeps the software and hardware both very simple.

We also need VADDU and VSUBU instructions. When src operand sizes are smaller than the destination, the smaller element sizes must be extended using zero-extension or sign-extension (VADDU, VADD). Perhaps some other instructions also need this signed-type information?

Comparison to current vector spec

From what I see, changing vsew realigns all of the named vector registers. Essentially, each named vector register is a pointer into an address space of the vector data store; changing vsew reassigns all of these address pointers. This is a difficult thing to implement in hardware, as it requires a few shifters to adjust the position of the address bits (and insert 0s into the LSBs); these shifters need to support shifting by 0(byte) to 4(quad) to 6(quadword register, vsew=doubleword) bit positions; the amount of the shift is determined by vsew and whether the register is configured to be a single, double or quad. This logic is fairly complex (but fortunately it is static for a given vconfig state). Because of differences in possible implementations, the current spec also requires zeroing all data elements because these pointers get reassigned, making it difficult for software to understand the data layout.

In the new proposal, each named vector register points to the same fixed address in the vector data store. The only thing that changes is how element data is interpreted within that vector data store; is it treated as bytes, halfwords, words, or doublewords. Even when the number of named vector registers are halved (and maxvl is extended), it merely removes references to intermediate access points (eg, through the odd numbered vector registers), so the named vector register pointers still all stay fixed. Since the data layout changes are readily transparent to software, there is no need to zero the content when the access element size changes.

The current use of vsew, and double/quad element sizes, is very flexible. All of that could be retained as part of the VTypes proposal, since there the element data type information must already be stored separately from the instruction encodings.

What I'm proposing may involve more work in redesigning the ISA, but that would only be "done once" and in advance. It would result in a lower cost to all implementations.

chenwei-tw, 2018-11-03:
Any discussion on this issue in V task group meeting this week?

rkruppe, 2018-11-03:
We didn't get around to it, earlier items on the agenda took too long.

vbx-glemieux, 2018-11-04:
One point of concern for my proposal above is differing MAXVL for 8, 16, and 32b data. This is difficult for software to track, and makes it difficult to combine vectors of differently sized elements (since v0 interpreted as 8b has twice as many elements as v1 interpreted as 16b).

One easy solution is to force 16b mode only access even-numbered registers; register pairs are bonded together to make the same MAXVL as 8b registers. That is, v0 and v1 would both have the same MAXL when accessed as 8b, but only v0 could be accessed as 16b and but it would combine data with v1 to produce the same MAXL (whether v1+v2 can be accessible in 16b mode can be debated). The drawback with this is there can would be, for example, 32 registers in 8b mode, 16 registers in 16b, mode, and only 8 registers in 32b mode. This restriction is similar to (but not as restrictve as) the current proposal when vsew is set to 8b and uses vconfig to assemble 16b and 32b registers, but the new proposal is more flexible because a set of registers can be interpreted as double or quad at any time without requiring a vconfig.

Another solution is to burden the programmer with the problem that 16b vectors can only be half as long as 8b vectors, etc. This involves using vsetvl and specifying which pair of data element sizes will be used (eg:   vsetvl rd, rs1, 16 would return the MAXVL whenever the largest operand size is 16b). The downside of this approach is that operations on 8b data with 16b data can only operate on half of the total possible elements in the 8b vector (eg, the first half of the elements). Accessing the other half would require a vslide of the 8b vector, or operations involving mixed-precision will always just be shorter.

There are other solutions -- eg, the solution with VectorBlox MXP is actually very elegant -- but that requires further architectural changes that I don't think the RISC-V Vector community is ready to consider.

rkruppe, 2018-11-04:
Having only 8 32 bit registers is already a serious problem, and extending that to 64 bit with only 4 registers sounds completely unacceptable for many domains -- even more so when you subtract one from each of these to account for the fact that v0 is needed for the mask. The architectures I know of that use aligned register groups to create larger registers for wider operations start from a much more reasonable basic size (e.g., 32b, combining pairs into 64b registers) and when they need to combine 4 or 8 registers it is only for some specific operations (e.g., texture sampling) and from a register field with far more than 32 registers.

> This restriction is similar to (but not as restrictve as) the current proposal when vsew is set to 8b and uses vconfig to assemble 16b and 32b registers, but the new proposal is more flexible because a set of registers can be interpreted as double or quad at any time without requiring a vconfig.

This is an unfair comparison: if you're operating on 64b data exclusively, you can set vsew to 64b and use all 32 registers. Even if you have some 8b data sprinkled in between, you have the option of storing that in a 64b-element register (with smaller MAXVL obviously), or you can make a trade-off and set vsew to something between 8b and 64b to get longer MAXVL at the cost of fewer 64b registers. The best choice here will depend on many factors but one of these choices will likely be better than having only three 64b registers even in mixed-precision workloads.

> Another solution is to burden the programmer with the problem that 16b vectors can only be half as long as 8b vectors, etc. This involves using vsetvl and specifying which pair of data element sizes [...]

These downsides are often unavoidable. The example you gave on the mailing list is very interesting but doesn't change that a lot of other algorithms can only process the same number of elements at each data width, so you need to either spread the vectors of larger element widths across multiple registers (and replicate instructions and move data around spuriously) or limit your parallelism (Arm SVE also has support for this FWIW, quite elegantly so by leaning on predication). This is pretty inherent in measuring registers in bits instead of elements.

kasanovic, 2018-12-01:
Closing for now, as new spec v0.6 addresses is new design point that addresses some of these concerns.

closed 2018-12-01 by kasanovic