riscvarchive/riscv-v-spec #65: example code for accessing other elements

brucehoult, 2018-12-02:
Could you provide an example for how you'd implement something like ...

void smooth(float *dst, float *src, int len){
  dst[0] = src[0];
  for (int i=1; i<len-1; ++i){
    dst[i] = (src[i-1] + 2*src[i] + src[i+1])/4;
  }
  dst[len-1] = src[len-1];
}

Or maybe:

void smooth(float *dst, float *src, int len){
  for (int i=0; i<len; ++i){
    dst[i] = (src[max(i-1,0)] + 2*src[i] + src[min(i+1,len-1)])/4;
  }
}

I'm assuming you'll want to use slides to get things in the correct lanes,  and merges to transfer values between iterations?

brucehoult, 2018-12-02:
It seems to me that using slides and merges will efficiently work to reference elements in other loop iterations that are constant offsets of less than the vector register length away.

But you don't know the vector register length when you compile the code. And it's legal to have the vector register length equal to 1 element.

aswaterman, 2018-12-02:
Great reason to mandate the implementation of the `vlmul` feature... :-)

aswaterman, 2018-12-02:
Can also just use multiple vector loads, offset one element from each other.  Simpler codegen, and might perform better or worse than slide, depending on lots of factors.

brucehoult, 2018-12-02:
Ah, ok, yes since loads don't need to be aligned you can just use a load for each offset. Maybe hardware can even optimise for that?

That solves this case. But not recurrences, where one iteration uses a value that was modified by an earlier iteration.

vbx-glemieux, 2018-12-02:
vslide to move data down
vinsert to add a new element at the end

i agree that an architectural maxvl of 1 could cause lots of problems with
code portability. it’s an unexpected case because slides and such break
down.

i think we should suggest a minimum of 4 elements as the architectural
maxvl. this should not overly emburden small implementations. i believe a
value of 2 is still too small here, as it will be difficult to accelerate
most code with a maxvl of 2.

note that software can still use an AVL of less than 4, and work properly
for those cases (eg, the last few cases after stripmining).

but software that expects to use vectors should never be told by the vector
engine that the maxvl is 1, as testing those cases is unnecesssrily
burdensome.

guy

vbx-glemieux, 2018-12-02:
> I hear you, but I think constraints on VLEN/ELEN need to be left out of
> the ISA spec and instead set by a given hardware platform specification.
>
> VLEN=ELEN (i.e. MAXVL=1) is a reasonable design point for some very
> low-cost implementations: you can still accelerate loop nests that need
> only 4-8 architectural registers, as VLMUL gets you MAXVL=8/4 in those
> cases.

i wouldn't consider such an ultra-low-cost implementation part of the V
spec.

i would call it a derivative that is non-compliant.

why?

because if you allow it as part of the V spec, then all good software needs
to be written for the case MAXVL=1. this means things like slides won't
work at all, so you need to provide an entirely different routine so that
you software works on such deviant architectures.

i'd rather allow MAXVL=1 as non-compliant versions, and software that wants
to work on those implementations can be specifically tested to run on
non-compliant hardware.

guy

vbx-glemieux, 2018-12-02:
it's not just manually written software ... but all compiler-generated code
would need to special-case MAXVL=1 and this will lead to code bloat.

guy

rkruppe, 2018-12-04:
It doesn't always help, but in the 0.6 scheme, one can infer a little more information about VLMAX than just that it's > 1. For example, on an ELEN=64 machine (e.g. RV64GV in a typical profile):
 
- For SEW=32 (e.g. as in @brucehoult's initial examples), VLMAX is a multiple of two
- For SEW=16, VLMAX is a multiple of four
- For SEW=8, VLMAX is a multiple of eight

These facts will also be known to autovectorizers.

kasanovic, 2019-02-22:
For Bruce's original case, the multiple loads from memory will probably be fastest on most implementations.  The other comments are regarding platform profile constraint on minimum VLMAX, and that is much broader topic.  Closing this for now.

closed 2019-02-22 by kasanovic

brucehoult, 2019-02-22:
To address @vbx-glemieux points, with the current design I think 32 bits per vector register is a reasonable design point for small implementations.

VLEN=32 and VLMUL=4 gives exactly the same configuration as ARM's new MVE, just with your eight vector registers using up five bits each in the instruction encoding :-)