riscvarchive/riscv-v-spec #66: SEW and VLMUL data layout patterns

vbx-glemieux, 2018-12-02:
I have a concern over the data layout being proposed, both when vector registers are lengthened with VLMUL and when the the standard element width SEW changes.

Any data layout must be easily understandable and usable by the programmer.

### VLMUL

When changing vlmul, it is not possible to "append" a vector register with more data because the values are interleaved.

Consider this example from the spec:

```
   Layout of element when vlmul=01, SEW=32b

        ELEN unit        3       2       1       0
        Byte          3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0
   v0                       3       2       1       0
   v16                      7       6       5       4

   Layout of element when vlmul=10, SEW=32b

        ELEN unit        3       2       1       0
        Byte          3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0
   v0                       3       2       1       0
   v8                       7       6       5       4
   v16                      B       A       9       8
   v24                      F       E       D       C
```

If software starts with vlmul=01, then goes to vlmul=10, the layout of the vectors changes because data in v8 now inserts itself in between the data contributed by v0 and v16. That is, the original data elements are now non-contiguous.

Instead, I suggest using adjacent vector registers to extend the length. As a side effect, this makes address generators that walk through an entire vector much simpler because vector data can be stored contiguously inside a RAM. 

Here is the how I propose the new layout should look:

```
   Layout of element when vlmul=01, SEW=32b

        ELEN unit        3       2       1       0
        Byte          3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0
   v2*n                      3       2       1       0
   v2*n+1                    7       6       5       4

   Layout of element when vlmul=10, SEW=32b

        ELEN unit        3       2       1       0
        Byte          3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0
   v4*n                     3       2       1       0
   v4*n+1                   7       6       5       4
   v4*n+2                   B       A       9       8
   v4*n+3                   F       E       D       C
```

### SEW

Likewise, when vector standard element size is increased, the current proposal
is to insert other bytes in between the smaller-sized element bytes.
It would be more natural to treat it like "zero extension" when
widening, and "truncation" when narrowing. This is natural for
programmers to understand, and results in a simple way to "prepare"
the data for the size adjustment from the programmers perspective.

Here is the current spec on double- and single-wide data layout:

```
 Example 1: ELEN=32
 ELEN unit      3       2       1       0
 Byte        3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0
 SEW=16b       7   6   5   4   3   2   1   0   <=16-bit elements
 v2*n              6       4       2       0   32-bit elements
 v2*n+1            7       5       3       1''
```

Notice that when reading any register as 8b elements (SEW=8), it is natural to have bit positions 7 to 0 of the vector store and ALU datapath to hold the data of its first element (element 0). Likewise, because of parallel SIMD operation, we probably want the bits of element 1 to be located at position 15..8.

However, after changing SEW to 16, the current proposal requires shifting bits 7..0 of v2*n+1 to bit positions 15..8. This is not a natural position for those bits. Likewise, it requires shifting bits 15..8 of v2*n down to position 7..0 so that it can be accessed as the LSB of element 1. Not only that, but if the overall SIMD width is only 16b, then these bits 15..8 of v2*n must be delayed until the next cycle of data is accessed, along with bits 15..8 of v2*n+1, to form the full 16b of element 1.

When changing SEW to 32, things follow a different pattern altogether. If a software programmer were to attempt to utilize the underlying data layout, the data shuffle pattern depends upon the source/dest values of SEW, so software would have to be rewritten for every case. With my proposal it's much simpler and a single routine can handle all cases.

Instead, I propose the following data layout:

```
 Example 1: ELEN=32
 ELEN unit      3       2       1       0
 Byte        3 2 1 0 3 2 1 0 3 2 1 0 3 2 1 0
 SEW=16b       7   6   5   4   3   2   1   0   <=16-bit elements
 v2*n              3       2       1       0   32-bit elements
 v2*n+1            7       6       5       4''
```

This is compatible with the VLMUL new data layout. For example, after performing a widening operation with vlmul=00, the first half of elements are stored in v2*n and second half stored in v2*n+1. Thus, changing vlmul=01, the same data layout is present, and the entire double-length vector can be accessed.

### Microarch implications

From a microarch standpoint, it's not clear how the current proposal helps to reduce datapath wiring. I would like to see an explanation of how it simplifies things -- it's not immediately clear to me. Such an implementation must be made public and not encumbered by any patents.

My proposal has a clear justification and explanation why it would simplify datapath wiring as well as address generation to walk through multiple values for different data widths and longer vectors. We have a working FPGA implementation of my proposal that proves this system is simple to implement. We have also implemented all forms of mixed-data operations (eg, widening adds) using this data layout.

brucehoult, 2018-12-03:
My guess about how this might work is as follows.

I assume that at some point the address of a byte in a vector register can be viewed as a concatenation of the register number and the byte within the register, in some order: reg||byte, or byte||reg.

I believe @kasanovic's proposal would correspond to byte||reg as follows (assuming 16 bytes in a single length register):

vlmul | byte | reg
--------|-------|-----
00 | bbbb | xyzrr
01 | bbbbx | yzrr
10 | bbbbxy | zrr
11 | bbbbxyz | rr

On the other hand @vbx-glemieux's proposal would look like:

vlmul | byte | reg
--------|-------|-----
00 | bbbb | xyzrr
01 | xbbbb | yzrr
10 | xybbbb | zrr
11 | xyzbbbb | rr

Is this a correct interpretation?

kasanovic, 2019-02-22:
This was heavily revised in 0.7 - please comment on new layout proposal.

closed 2019-02-22 by kasanovic