riscvarchive/riscv-v-spec #69: Question about the encoding of vector load/store

rofirrim, 2019-01-07:
Hello,

I'm a bit confused with the exact encoding of the vector load and stores in 0.6.

§13.2 of the draft spec describes the encoding of loads and stores. For this issue I will focus on loads but I face a parallel issue with stores.

If I understand the text correctly in §13.2, a vector load is encoded like shown below:

```
Inst{31-27} = uoffset5;
Inst{26-25} = {m1, m0}; // vector mask
Inst{24} = xv;     |
Inst{23} = order;  |  this is is the funct5 field
Inst{22-20} = mop; |
Inst{19-15} = base-register;
Inst{14-12} = width; // one of the ones reserved for the V-extension
Inst{11-7} = destination-register;
Inst{7-0} = LOAD-FP // standard opcode
```

My question is, how is the second source operand (`rs2` / `vs2`)encoded in the constant-stride instructions (§13.3.2) and in the indexed instructions (§13.3.3)?

Assuming my understanding of the encoding, as shown above, is correct, then my best bet at this point is that when `mop` encodes a constant-stride or indexed memory access then `uoffset5` encodes a X or V register. If this were the case then constant-stride and indexed memory accesses wouldn't have an offset, not sure if this was the intent.

The instruction table document shows an encoding that, as far as I understand, does not match the text of the spec. So perhaps the text of the draft spec needs updating instead?

Thank you very much,

closed 2019-01-08 by colinschmidt

colinschmidt, 2019-01-08:
You were correct the text needed updating.
Thanks for the report.

rofirrim, 2019-01-08:
Thank you very much Colin!

commit 8d9c8b01 (2019-01-08): Update first half of load/store description

This matches the change made recently to recover
immeadiates for non-unit-strided load/stores
Fixes #69