riscvarchive/riscv-v-spec #83: EDIV vsetvli syntax

jim-wilson, 2019-02-13:
The EDIV section doesn't clearly specify the strings the assembler should use with vsetvli for setting the vediv field.  One code example uses "vdiv4" and another code sample uses "d4".  The "d4" is more consistent with the "e8" and "m4" documented in the vsetvli section, so I'd suggest dX for X=1,2,4,8.

One code sample also uses "ve32" should should be "e32".

kasanovic, 2019-02-22:
Couldn't find these now, so must have been fixed meantime.

closed 2019-02-22 by kasanovic

jim-wilson, 2019-02-22:
I saw the git log entry that mentioned that dX was the new syntax, and so submitted pull requests to fix assembly code that Andrew merged in.

However, we are still missing a table to document the assembly syntax.  Section 6.1 has a table "Suggested assembler names used for vtypei setting" that documents the eX and mX values for vsetvli, but the dX values for the Zvediv extension are not documented anywhere, they are just used in a few assembly code samples.

kasanovic, 2019-02-23:
Added clarification.

Krste

| I saw the git log entry that mentioned that dX was the new syntax, and so submitted pull requests to fix
| assembly code that Andrew merged in.

| However, we are still missing a table to document the assembly syntax. Section 6.1 has a table "Suggested
| assembler names used for vtypei setting" that documents the eX and mX values for vsetvli, but the dX values for
| the Zvediv extension are not documented anywhere, they are just used in a few assembly code samples.

| —
|