riscvarchive/riscv-v-spec #109: example algoritm problematic

David-Horner, 2019-03-05:
the narrative for **13.5. Vector Single-Width Scaling Shift Instructions** clearly defines the function, however the formula in the examples does not explicitly reference the sign/zero extension. Rounding that would carry into or our of the high order bit are problematic.

consider SEW=8:
 value in the element X'FF' vssrl by 1 should be X'80' , formula yields X'00'
 value in the element X'7F' vssra by 1 should be X'40' , formula yields X'C0'

I couldn't come up this an obviously good notation to express the required value extension but perhaps this will do:

 # For vxrm=rnu, round = 1 << (src2-1)
 # Scaling shift right logical
 vssrl.vv vd, vs2, vs1, vm   # vd[i] = (zero_extend(vs2[i]) + round)>>vs1[i]
 vssrl.vx vd, vs2, rs1, vm   # vd[i] = (zero_extend(vs2[i]) + round)>>x[rs1]
 vssrl.vi vd, vs2, imm, vm   # vd[i] = (zero_extend(vs2[i]) + round)>>imm

 # Scaling shift right arithmetic
 vssra.vv vd, vs2, vs1, vm   # vd[i] = (sign_extend(vs2[i]) + round)>>vs1[i]
 vssra.vx vd, vs2, rs1, vm   # vd[i] = (sign_extend(vs2[i]) + round)>>x[rs1]
 vssra.vi vd, vs2, imm, vm   # vd[i] = (sign_extend(vs2[i]) + round)>>imm

. Vector Single-Width Scaling Shift Instructions

These instructions shift the input value right, and round off the shifted out bits according to vxrm. The scaling right shifts have both zero-extending (vssrl) and sign-extending (vssra) forms. The shift amount can come from a vector or a scalar x register or a 5-bit immediate. The low lg2(2*SEW) bits of the vector or scalar shift amount value are used (e.g., the low 6 bits for a SEW=64-bit to SEW=32-bit narrowing operation). The immediate form supports shift amounts up to 31 only and is effectively zero-extended for SEW of > 32.

 # For vxrm=rnu, round = 1 << (src2-1)
 # Scaling shift right logical
 vssrl.vv vd, vs2, vs1, vm   # vd[i] = ((vs2[i] + round)>>vs1[i])
 vssrl.vx vd, vs2, rs1, vm   # vd[i] = ((vs2[i] + round)>>x[rs1])
 vssrl.vi vd, vs2, imm, vm   # vd[i] = ((vs2[i] + round)>>imm)

 # Scaling shift right arithmetic
 vssra.vv vd, vs2, vs1, vm   # vd[i] = ((vs2[i] + round)>>vs1[i])
 vssra.vx vd, vs2, rs1, vm   # vd[i] = ((vs2[i] + round)>>x[rs1])
 vssra.vi vd, vs2, imm, vm   # vd[i] = ((vs2[i] + round)>>imm)

aswaterman, 2019-11-29:
The text has been changed to use the `roundoff` function, but this issue persists.  We should perhaps define `roundoff_unsigned` and `roundoff_signed` functions.  I'll make a PR.

David-Horner, 2019-12-03:
Looks good to me.
The roundoff text is much better than original.
And now comprehensive with the signed/unsigned .
Thanks.

aswaterman, 2019-12-03:
Cool, thanks for the review.

PR #343 (2019-12-03): Distinguish signed and unsigned roundoff functions
Resolves #109
commit 9273cfde (2019-12-03): Distinguish signed and unsigned roundoff functions (#343)

Resolves #109
closed 2019-12-03 by aswaterman