riscvarchive/riscv-v-spec #114: Segment Load / Store assembler mnemonics. Make `nf` field an immediate argument

drom, 2019-03-13:
Section 7.8.{1,2,3} "Vector ---- Segment Loads and Stores"

Assembler mnemonics for vector Load/Store include `<nf>` field in the body.

Is it possible to put `<nf>` as a separate immediate argument?

IMHO, it would greatly simplify Assembler mnemonic if instruction like:

```
vlxseg7bu.v v4, (x5), v3
```

Will be written as:

```
vlxsegbu.v v4, (x5), v3, 7
```

PkmX, 2019-04-11:
I agree. This should also simplify assembler implementation.

Another point: should `nf == 1` be allowed as an alias to the non-segmented version?

kasanovic, 2019-04-19:
Decided to stay with current PoR, since immediate affects how many destination vector registers are written and different variants might be implemented differently in hardware.

closed 2019-04-19 by kasanovic