riscvarchive/riscv-v-spec #116: Non-resumable operations

go-sc, 2019-03-15:
> A few vector instructions can not be executed with a non-zero vstart value and will raise an illegal instruction exception as defined below.

As far as I can see, these instructions are not resumable, i.e. they raise an illegal instruction exception when `vstart` is 0:

* all reduction operations
* `vmpopc`
* `vmfirst`
* `vmiota`
* `vcompress`

Could you put more details on why exactly these instructions cannot be continued?
In particular, restarting reduction operations on long vectors can be harmful for performance.

kasanovic, 2019-03-31:
For these instructions, there is no obvious place to put the partially completed reduction result to allow the instruction to be restarted.   For the vmpopc/vmfirst instructions, the destination x register might be considered an obvious place, but that would require pipelines to allow x registers to be modified by instructions that cause exceptions - which is not otherwise required and might be very difficult to support in machines with scalar register renaming.  For vmiota, the instruction would have to be defined to first read one value from the destination register if vstart is non-zero, which is possible but adds complexity to the definition and implementation of this instruction.  Vcompress would also require some new additional state to record the destination element index.

These operations are all register-register, so restarting will only happen for asynchronous interrupts.  The only reason to interrupt in middle of these instructions is to reduce interrupt latency.  In the case that interrupt latency is important and the occupancy of these instructions is too long,  restart could be supported with a non-precise implementation-specific restartable interrupt mechanism having additional state.  For unix-class systems, it is unlikely the interrupt latency is a real issue.

closed 2019-05-30 by kasanovic

David-Horner, 2020-01-21:
vmfirst is an interesting exception to the others as vstart itself is the additional state information required to resume.
If the first set bit was found at point of interrupt then rd can be written and the next instruction be the *epc value.
If the first set bit was not found at point of interrupt but all vl bits were checked then rd can be written with -1 and the next instruction be the *epc value.
If the first set bit was not found at point of interrupt and all vl bits were not checked then rd cannot be written and vstart can be set to <= number of processed bits.

I expect vstart in the last case would be generally be set to n * SLEN.

Further, interrupts during vmfirst can be expected to have a greater frequency in fused and chained implementations.
Consider the situation of vmfirst processing the results from a preceeding vector compare instruction.
An implementation could process the two in parallel (and even earlier vector instructions). 
The vector compare could reasonably complete  while processing the interrupt (as with a divide instruction). 
But the vmfirst cannot predictably complete before the vd register is referenced nor the vstart CSR be read.
Hence, a higher incidence of interrupts "occurring" during vmfirst.
 
I intend to recommend that other vstart=0 exception constraints be relaxed.

And to generalize the " restart ... supported with a non-precise implementation-specific restartable
 interrupt mechanism" and make it the default.