riscvarchive/riscv-v-spec #117: Bad asciidoc format

aka-sps, 2019-03-20:
After #112 *.adoc files are not asciidoc files (JSON patches violates asciidoc format)

gurjeet, 2019-03-20:
Cross posting here; original: https://github.com/riscv/riscv-v-spec/commit/7dfbd2735763998ee2e391abd432842e9ae3d5a0#commitcomment-32835055

@aswaterman @drom it feels like allegations made by @aka-sps are correct, and I'd side with him.

The resulting docs are not proper AsciiDoctor formatted and the repo holding the build infrastructure is not part of RISC-V GitHub Org. I don't have a way to verify if these build scripts are actually being used to produce the official versions of the docs; as of now these are hosted on https://drom.io/vector/.

This smells of a bad policy, and prevents readers and users of this repo from using it without also depending on @drom's repo.

aswaterman, 2019-03-20:
Temporarily resolved by #118.  To be clear, we are going to adopt @drom's approach, but we should resolve these concerns first.

closed 2019-03-20 by aswaterman

gurjeet, 2019-03-20:
@aswaterman Thanks!

drom, 2019-03-22:
@aka-sps @gurjeet Addressing issues above I have published:
* document generation setup here: https://github.com/riscv/riscv-v-spec/blob/master/package.json
* Travis CI config to publish HTML/PDF on GitHub pages: https://github.com/riscv/riscv-v-spec/blob/master/.travis.yml
* Deployed HTML: https://riscv.github.io/riscv-v-spec/
* Deployed PDF: https://riscv.github.io/riscv-v-spec/v-spec.pdf

drom, 2019-03-22:
> After #112 *.adoc files are not asciidoc files (JSON patches violates asciidoc format)

Could you be more specific about incompatibilities?
I am running `asciidoctor v-spec.adoc`  with no issues.

aka-sps, 2019-03-22:
> I am running `asciidoctor v-spec.adoc` with no issues.

~~~~
asciidoctor-pdf --trace -o v-spec.pdf v-spec.adoc
/home/user/.gem/ruby/2.3.0/gems/prawn-2.2.2/lib/prawn/text/formatted/line_wrap.rb:243:in `update_output_based_on_last_fragment': Prawn::Errors::CannotFit (Prawn::Errors::CannotFit)
~~~~

drom, 2019-03-22:
> > I am running `asciidoctor v-spec.adoc` with no issues.
> 
> From [af27fea](https://github.com/riscv/riscv-v-spec/commit/af27fea5305b9d703ab57b9d3b2a8638cfdc984a)
> 
> ```
> asciidoctor-pdf --trace -o v-spec.pdf v-spec.adoc
> line_wrap.rb:243:in `update_output_based_on_last_fragment': Prawn::Errors::CannotFit (Prawn::Errors::CannotFit)
> ```

I am not familiar with your generation flow. Could you track down what exactly in `*.adoc` causing this error? How critical is an error? Does it produce PDF?

drom, 2019-03-22:
> ```
> asciidoctor --backend html   --base-dir ./ --source-dir ./ -o v-spec.html v-spec.adoc
> ```
> Result is (fragment):
> 
> ```
> vtype register layout
> 
> {isa: [
>   {bits: 2,  name: 'vlmul', attr: 'RW'},
>   {bits: 3,  name: 'vsew', attr: 'RW'},
>   {bits: 2,  name: 'vediv', attr: 'RW'},
>   {bits: 26},
> ], config: {bits: 16}}
> ```
> And how did you test?

The fact that your documentation flow doesn't render ISA descriptor as an Image does not make it "Bad asciidoc format" ?