riscvarchive/riscv-v-spec #119: Do not put pdf files under version control

go-sc, 2019-03-22:


drom, 2019-03-22:
Travis CI flow with deployment to the Github pages was introduced to create HTML / PDF files from the specs merged to the `master` branch. By default, Github pages service stores all files in `gh-pages` branch. I don't know a better, opensource, free and automatic way of releasing PDF documents. Any suggestions?

aswaterman, 2019-03-22:
We could do it the way I did for riscv-isa-manual, where deployment creates a release, but does not push to a branch.

drom, 2019-03-22:
But then HTML/PDF will not be published on the Web server.

go-sc, 2019-03-22:
GitHub releases feature is suitable for publishing binary files for (sort of) stable releases.
Anyone willing to create a pdf file for a certain commit can do it with a simple makefile that calls Asciidoctor with proper arguments.
HTML version of .adoc files generated by GitHub is OK, so probably there was no need in continuous integration at all.

drom, 2019-03-22:
I have removed Travis-CI config, flow and `gh-pages` branch. https://github.com/riscv/riscv-v-spec/commit/f51e1b315fd63053942744063087f729ada49fbd
HTML / PDF are not stored in this repo.

go-sc, 2019-03-22:
Yes, closing.
Thank you.

closed 2019-03-22 by go-sc