riscvarchive/riscv-v-spec #121: inconsistent naming of vredsum.vs

chihminchao, 2019-03-26:
[15.1](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#151-vector-single-width-integer-reduction-instructions) defines the vector reduction as **vredsum.vs** but [19.2](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#151-vector-single-width-integer-reduction-instructions) describes it as **vredsum.vv**.

Should **vredsum.vs** be preferable one ?

gurjeet, 2019-03-31:
@kasanovic, @colinschmidt, pinging you as you guys are major contributors here. The complaint by @chihminchao sounds legitimate.

There's a copy-pasta error with his second link, though. The correct links are: [15.1][1] [19.2][2]

[1]: https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#151-vector-single-width-integer-reduction-instructions
[2]: https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#192-instructions-affected-by-ediv

aswaterman, 2019-03-31:
@gurjeet they aren't ignoring these issues; they are just busy.  If you want to speed things up, make a pull request that addresses the issue.

closed 2019-03-31 by kasanovic

gurjeet, 2019-04-01:
> @gurjeet they aren't ignoring these issues; they are just busy. If you want to speed things up, make a pull request that addresses the issue.

I'm pretty sure they are. I wanted to bubble up the OP's concern since it had been a few days since complaint. Next time I'll make a PR for such minor complaints instead of writing a comment :)

commit 697b987d (2019-03-31): Fixed typo vredsum.vv -> vredsum.vs
Closes #121