riscvarchive/riscv-v-spec #130: can vector instruction with vl==0 be dropped completely as NOP?

solomatnikov, 2019-04-16:
E.g. vector load can cause a PMP or page table exception even if it doesn't update any register state.

Also, instructions can update stats counters and have other side effects.

aswaterman, 2019-04-16:
According to the current spec, even if vl=0 (or vstart >= vl), the elements at vl and beyond still need to be written with zero.

But no memory should be accessed, so there should be no page faults or access exceptions.

solomatnikov, 2019-04-16:
https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#vector-length-register-vl

> As a special case, when vl=0, no elements are updated in the destination vector.

This sentence says no elements are updated when vl==0

aswaterman, 2019-04-16:
Ah, my mistake, what I said is only true for the vstart >= vl case.

vbx-glemieux, 2019-04-16:
what if vl=0 and vstart=0 ?

(sorry not trying to be pedantic, but this seems to be a plausible case
that has ambiguity right now)

guy

aswaterman, 2019-04-16:
Sidestepping your question, I'm wondering what the merits are of treating vstart >= vl differently than vl = 0.  An alternative is to keep all elements intact whenever vstart >= vl, which would subsume the vl = 0 case.  Not sure what the downsides might be.

kasanovic, 2019-04-19:
Closed by pull request #133.

closed 2019-04-19 by kasanovic