riscvarchive/riscv-v-spec #132: imm value range of vsleu.vi

davetw, 2019-04-16:
In 12.7 note.
> The vsleu.vi range is 0 to 15 (and (~0)-15 to ~0)

Does it mean the value range of imm is 0 to 15? If so, why only use 4 bits instead of 5 bits?

aswaterman, 2019-04-16:
`0 to 15 (and (~0)-15 to ~0)` means `0 to 15 and 2^N-16 to 2^N-1`, where N is the width of the argument.  It's 32 distinct values, hence 5 bits.  I believe it's written this way because the arguments are unsigned, and so writing `-16 to 15` would be confusing in a different way.

kasanovic, 2019-04-19:
The immediate is sign-extended to SEW bits in the same way for all instructions, which leads to some of the unsigned operations having less useful ranges in some cases.

closed 2019-04-19 by kasanovic