riscvarchive/riscv-v-spec #137: What are the reset values of vl, vtype, etc?

solomatnikov, 2019-04-18:


aswaterman, 2019-04-18:
My opinion is that, as long as vtype.vill is 1 after reset, the other fields and registers don’t matter, and can even be inconsistent (e.g. vl > MAXVL). If vtype.vill is 1, then the vector unit won’t be usable until vsetvl is executed, which will bring the system into a consistent state.

kasanovic, 2019-04-19:
I agree with view that vtype.vill should be set to 1 on reset.

kasanovic, 2019-04-19:
Maybe could be weaker requirement that state has to be a supported vtype, or vtype.vill=1.

aswaterman, 2019-04-19:
Weaker requirement seems OK (so either vill=1 and all other state is unspecified and possibly inconsistent, or vill=0, vtype supported, and state is consistent, i.e. vl <= VLMAX).

commit 9bffbca4 (2019-04-21): Defined reset values for vector extension.
Closes #137.
closed 2019-04-21 by kasanovic