riscvarchive/riscv-v-spec #139: 16.1: Vector Mask-Register Logical Instructions: bitwise or mask-element-wise?

jsgray, 2019-04-19:
I could not tell from the spec / example notation whether mask-register logic instructions operate bitwise or mask-element-wise.

For example, when SEW=8 LMUL=1 => MLEN=8, and if VLEN=64, and if we have
v1 = 0x88.77.66.55.44.33.22.11 (as a mask: 0.1.0.1.0.1.0.1)
v2 = 0x0f.0f.0f.0f.0f.0f.0f.0f (as a mask: 1.1.1.1.1.1.1.1)
after vmand.mm v3,v1,v2
is v3 == 0x08.07.06.05.04.03.02.01 or 0x00.01.00.01.00.01.00.01?

aswaterman, 2019-04-20:
The pseudocode (e.g. `vd = vs2 & vs1` for `vmand`) suggests a bitwise operation, which seems like the right definition to me.  The alternative mask-element-wise variant that produces only 0 or 1 is more expensive to implement but doesn't seem to benefit software.  I'll PR a clarification, and the others will let us know if I'm wrong :)

kasanovic, 2019-04-21:
It was intended to be bitwise to reuse the hardware for logical operations, and as Andrew points out, software should not particularly care.

PR #142 (2019-04-21): Clarify that mask instructions operate bitwise within a mask element
At least, I think it's a clarification, not a change...

Closes #139
commit 0c9d6b99 (2019-04-21): Mask instructions operate bitwise within a mask element

Closes #139
closed 2019-04-21 by kasanovic