riscvarchive/riscv-v-spec #147: Portability of changing LMUL

rednoah91, 2019-04-26:
I tried to increase LMUL in order to gain performance, for example e32 m1 -> m8.
The register allocation in the code need to be whole modified
Ex: rename v0, v1, v2 to v0, v8, v16

Why can't we keep the same register-name but different meaning after LMUL changed?
Ex: when e32 m1 -> m8, 
v0 means v0~v7
v1 means v8~v15
v2 means v16~v23
...
so that we don't need to rename the register.
How do you think about it : )?

jnk0le, 2019-04-26:
The masks are held in single register regardless of LMUL setting. So that approach would lead to very confusing code where you have data in v1, v2, ... and some set of masks in v1, v2, ...

solomatnikov, 2019-04-26:
Current scheme simplifies HW

kasanovic, 2019-04-29:
You can always keep the same register names if you always allocate in order v0, v8, v16, v24, v4, v12,...

I always use #define in cpp to simplify changing assembler register allocation when writing by hand.  Also allows use of more descriptive names, and also handles this issue.

closed 2019-04-29 by kasanovic