riscvarchive/riscv-v-spec #151: Confusion about assembly syntax pattern for vector binary arithmetic instructions

wwang0, 2019-04-29:
Current spec uses vs2 as the first operand and rs1/simm5 as the second operand. It is opposite to most of RV instructions, like standard scalar ordering, or vector ternary arithmetic instructions.  It does cause some confusion , and have you ever thought about using bit[19, 15] just for vs1, and bit[24, 20] for vs2, simm5 and rs2?  And in my view it is more compatible with standard scalar instructions both from encoding and assembly syntax perspectives. 

I know you want scalar register starts from rs1, but if you check current vector load/store instruction, there is no vs1 and we start directly from vs2;)

wwang0, 2019-04-30:
In the spec, I see expression like 'the first operand specifier (vs2)', or 'the second operand specifier(vs1)'  a few times, always feel awkward when I see it..

wwang0, 2019-04-30:
another example: vfsgnj vs fsgnj. vfsgnj will get the sign bit from first operand(vs2), but fsgnj will get it from second operand(rs2)....

wwang0, 2019-05-02:
And another reverse example: 
vfdiv.vf vd, vs2, rs1, vm
The result I guess is: vd[i] = vs2[i]/f[rs1]

Integer or FP:
fdiv.s rd, rs1, rs2
The result is rd = f[rs1]/f[rs2].

Here both vector and integer&f provide natural reading of the assembler(which is different from above example), but rs1 field does be used differently between v-spec and Integer&FP.

kasanovic, 2019-05-03:
Using " bit[19, 15] just for vs1, and bit[24, 20] for vs2, simm5 and rs2" would add new paths to scalar logic with only gain being some simplification in documentation (but even assembly programmers won't see the issue as they don't consider vs2/vs1 specifier names).

Current vector load/store instructions source rs1 for address in same way that scalar instructions source rs1 for address, and use vs2 for source in same way as arithmetic instructions.

closed 2019-05-03 by kasanovic

wwang0, 2019-05-04:
@kasanovic , I am not sure I quite understand "adding new paths to scalar logic", could you elaborate when you are free. Thanks.