riscvarchive/riscv-v-spec #154: vext.x.v / vmfv.f.s asymmetry

aswaterman, 2019-04-30:
The integer variant can extract any element, based upon `x[rs1]`, whereas the floating-point variant can only extract element 0.  These should probably be made symmetrical, either by changing the integer variant to only allow extraction of element 0, or by changing the floating-point variant to allow extraction of any element.

I favor the former option for its simplicity.  vext was provided largely for allowing a debugger to read out vector registers without touching memory, but a debugger can do that with vslide1down instead.  And in application code, vext can be emulated with vslidedown or vrgather with vl=1, followed by extracting element 0.

JerryShih, 2019-05-17:
Could we add the same function as vext for floating value?
How to emulate vext using vslidedown or vrgather? It's easier to use vext with your index value.

aswaterman, 2019-05-17:
Easier, yes, but not necessarily much faster.  In many implementations, vext will cause a loss of decoupling, which will have a more significant performance impact than the additional cost of the `vslidedown`.

aswaterman, 2019-07-09:
Related issue: although I still think we should redefine vext.x.v extract element 0 rather than element x[rs1], if we choose to go the opposite direction and allow extraction from any element position, we should consider redefining these instructions to operate on register groups instead of registers.

PR #247 (2019-07-15): Replace vext.x.v with vmv.x.s
This regularizes the integer moves with the floating-point ones, and simplifies the implementation.

The effect of vext.x.v can be obtained with vslidedown.vx followed by vmv.x.s.

Closes #154
commit 83fc2789 (2019-07-15): Replace vext.x.v with vmv.x.s

This regularizes the integer moves with the floating-point ones, and
simplifies the implementation.

The effect of vext.x.v can be obtained with vslidedown.vx followed by
vmv.x.s.

Closes #154
closed 2019-07-15 by kasanovic