riscvarchive/riscv-v-spec #155: Carefully consider vadc/vsbc vs. simpler alternative

aswaterman, 2019-05-01:
I think there's agreement on the need for performant multi-word addition support, and vadc does satisfy that need.  However, it's a wart, since it writes two vector register groups.  This is annoying across all implementation styles.  The problems are surmountable, but we should carefully evaluate whether it's worth the effort.

The most obvious alternative is to split it in two (i.e., compute vd = vs2 + vs1 + v0, and separately compute vd = carry(vs2 + vs1 + v0)).  This will also result in an extra mask-register move in some cases (namely, when vd overlaps vs1 or vs2), which can be removed at the cost of code size by unrolling the loop once.

ccelio, 2019-05-03:
I don't have strong feelings in either direction, but I would be okay with *keeping* the more complicated, two-destination version.

I think many cores will probably have to crack it into two separate uops, but this probably won't be the first instruction that needs cracking.

Although I'm a fan of simplicity when macro-op fusion can save the day, the extra moves make fusion much harder to catch.

aswaterman, 2019-05-03:
Unfortunately, this one is hard to split into uops if you don’t do renaming, for the same reason there is sometimes an extra move in the split variant: the instruction always overwrites its v0 input and sometimes overwrites its rd input, in which case there’s no way to order the two uops in a way that both have their operands preserved. To guarantee you can always do this, you’d have to require that rd not overlap rs1 and rs2, which sometimes adds back the extra move instruction, and this time it’s a register-group move, not just a mask move.

In my opinion, this makes the longer but fusable sequence preferable. The extra move doesn’t have to be part of the fusion sequence (it can be instruction #3), and not fusing the move isn’t much of a loss, since it only moves a mask, not a whole register group.

billhuffman, 2019-05-16:
I agree with Andrew that two instructions are better.  Having two destinations adds difficulties both in dispatch and in allocation of register file write ports.  I think following the same line of thinking that's done for scalar integer multiply - two instructions and a definition for what constraints software should follow to enable fusing the two - would be much better here.

aswaterman, 2019-05-17:
Per the last TG meeting, the two-instruction sequence is the new POR.  @kasanovic is going to push an update to the spec soon.

commit bc6639b1 (2019-05-31): Replaced adc/subc with two instructions each to separately produce sum
and carry values.
Closes #155
Closes #174
closed 2019-05-31 by kasanovic