riscvarchive/riscv-v-spec #159: vssrl&vssra vs vnclipu&vnclip

wwang0, 2019-05-02:
vnclipu&vnclip seems just be the narrowing version of vssrl&vssra, why don't we rename them as vnssrl&vnssra?   In my view, vnclipu&vnclip is more close to right shift instructions with saturating and rounding  than a conventional clip.

Thanks

chuanhua, 2019-05-03:
The main function of vnclipu and vnclip is the clipping part from both ends of a SEW-size-determined value range, unsigned and signed.  The term of "narrowing" does not convey how the "narrowing" operation is performed to reduce a 2*SEW-sized value into a SEW-sized value range. So I think the original names are appropriate.

kasanovic, 2019-05-03:
Thanks for explanation @chuanhua

closed 2019-05-03 by kasanovic

wwang0, 2019-05-04:
@chuanhua , @kasanovic   My point is using clip loses 'shift right' meaning. As for "The term of "narrowing" does not convey how the "narrowing" operation is performed to reduce a 2*SEW-sized value into a SEW-sized value range",  actually the first s(saturating and rounding) + n tells it. 

I know there is no perfect name, so the current names seems ok, but just share my opinions here.