riscvarchive/riscv-v-spec #160: Not all instructions that take mask register as destination operand are prefixed by vm*

wwang0, 2019-05-03:
a few instructions that take mask register as destination operand are not prefixed by vm*:
vseq, vsne, vsltu, vslt, vsleu, vsle, vsgtu, vsgt, vfeq, vfle, vford, vflt, vfne, vfgt.

To make names more meaningful, I suggest to rename them as:
vmseq, vmsne, vmsltu, vmslt, vmsleu, vmsle, vmsgtu, vmsgt, vmfeq, vmfle, vmford, vmflt, vmfne, vmfgt.

Then we have unified vector master instructions naming convention:
vmops.vv
vmops.vx
vmops.vi
vmops.vf
vmops.mm
vmops.m

wwang0, 2019-05-17:
@aswaterman , @kasanovic , any comments? thanks

aswaterman, 2019-05-17:
I agree with prefixing the comparisons with `vm`.

wwang0, 2019-05-18:
Thanks @aswaterman for your reply . and i will prepare a pull request for review.

closed 2019-05-31 by kasanovic