riscvarchive/riscv-v-spec #166: 3-operand opcodes order in specification

simon5656, 2019-05-09:
For all 3-operand opcodes, a toolchain from github that implements the vectors disassembles as
`<op> <vd>, <vs1>, <vs2>, <vm>`, but the specification 0.7 specifies 
`<op> <vd>, <vs2>, <vs1>, <vm>` 

Which is correct?

simon5656, 2019-05-10:
Jim Wilson of SiFive says:

There is an exception in the spec for this case.  This text was added Feb 23.

NOTE: For ternary multiply-add operations, the assembler syntax always
places the destination vector register first, followed by either `rs1`
or `vs1`, then `vs2`.  This ordering provides a more natural reading
of the assembler for these ternary operations, as the multiply
operands are always next to each other.

closed 2019-05-10 by simon5656

chihminchao, 2019-05-14:
@jim-wilson
Is it possible to change the ternary format from vd, vs1, vs2 to vd,  vs2, vs1 to align all the instruction syntax.  The exception makes code generation of test case or assembly programming complicated.  Code generation needs to swap the operands and programmer needs keep the difference in mind for these cases.

jim-wilson, 2019-05-14:
This was discussed back in February.  My initial binutils port had vs2 before vs1 and it was reported as a bug.  After the discussion, I had to fix binutils to put vs1 before vs2, and the docs were updated to clearly state that this is correct.  This is a little annoying, but it has the advantage that the order of the operations matches the order of the operations in the text.  With the other operand order, the text descriptions of the instructions is less clear.

chihminchao, 2019-05-15:
Could  you point out the discussion link in Feb ?

jim-wilson, 2019-05-15:
I looked for it but didn't find it.  It might have been an internal discussion.