riscvarchive/riscv-v-spec #172: Does the stride of vector-strided-segment-load/store instructions could be negative?

JerryShih, 2019-05-10:
In the [spec](https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc#782-vector-strided-segment-loads-and-stores), there is no restriction for the stride value.
Could the stride be negative? I think the spec should tell more about the stride value.
I have a case to access the memory from back to front.

Assume x6 = -3, Which pattern is used?
A or B?
a)
```
vlsseg3b.v v4, (x5), x6   # Load bytes at addresses x5+ i*(-3) into v4[i],
                          # and bytes at addresses x5+i*(-3)+1 into v5[i],
                          # and bytes at addresses x5+i*(-3)+2 into v6[i].
```
b)
```
vlsseg3b.v v4, (x5), x6   # Load bytes at addresses x5+ i*(-3) into v4[i],
                          # and bytes at addresses x5+i*(-3)-1 into v5[i],
                          # and bytes at addresses x5+i*(-3)-2 into v6[i].
```

aswaterman, 2019-05-10:
As you point out, there is no restriction on the stride.  A stride of -3 gives addresses (rs1, rs1 - 3, rs1 - 6, ...).  This doesn't require a modification to the normative text, but a note might be appropriate.

JerryShih, 2019-05-13:
@aswaterman 
I created a pr #173 to add a note for negative stride value.

aswaterman, 2019-05-13:
Closed via 9688e01837c512466ddd73ac9f79147a1fe1b2d8

closed 2019-05-13 by aswaterman