riscvarchive/riscv-v-spec #183: Cannot build documentation

eroom1966, 2019-05-20:
I am trying to rebuild the documentation from the latest *.adoc in the repository
if I follow the link to here from the README.md
https://riscv.github.io/documents/riscv-v-spec/riscv-v-spec.pdf
I think this is an older (cached) version ?

The problem I am having is that I get the following error message

> npm run build
> > riscv-v-spec@0.1.0 build /home/moore/git/riscv/riscv-v-spec
> > datasheet -i v-spec.adoc -o public/riscv-v-spec.pdf
> sh: 1: datasheet: not found

To the best of my knowledge my setup is correct, when I look in the detailed npm log I see the following

> 9 verbose lifecycle riscv-v-spec@0.1.0~build: CWD: /home/moore/git/riscv/riscv-v-spec
> 10 silly lifecycle riscv-v-spec@0.1.0~build: Args: [ '-c', 'datasheet -i v-spec.adoc -o public/riscv-v-spec.pdf' ]
> 11 info lifecycle riscv-v-spec@0.1.0~build: Failed to exec build script
> 12 verbose stack Error: riscv-v-spec@0.1.0 build: `datasheet -i v-spec.adoc -o public/riscv-v-spec.pdf`
> 12 verbose stack spawn ENOENT
> 12 verbose stack     at ChildProcess.<anonymous> (/home/moore/.nvm/versions/node/v12.2.0/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:48:18)
> 12 verbose stack     at ChildProcess.emit (events.js:196:13)
> 12 verbose stack     at maybeClose (internal/child_process.js:1011:16)
> 12 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:268:5)
> 13 verbose pkgid riscv-v-spec@0.1.0
> 14 verbose cwd /home/moore/git/riscv/riscv-v-spec
> 15 verbose Linux 4.15.0-47-generic

If I am doing something wrong, I'm not quite sure what it is.
Thx
Lee

jim-wilson, 2019-05-20:
Probably an issue with your local npm.  There is a much simpler way to build the docs by running asciidoctor.

eroom1966, 2019-05-22:
Thanks for the solution

eroom1966, 2019-06-07:
unfortunately asciidoctor throws the following error
Prawn::Errors::CannotFit
  Use --trace for backtrace#
adding --trace, simply outputs a huge backtrace (as expected) but does not help
anyone else using asciidoctor on this ?

aka-sps, 2019-06-07:
The source of problems with "Prawn::Errors::CannotFit" is manual "wrapped" layouts and columns width specifications of the tables in "inst-table.adoc"
Try removing the column width specifications:

~~~~
// [cols="4,1,1,1,8,4,1,1,8,4,1,1,8"]
~~~~

Premature formatting is as bad as premature optimization :-)

kasanovic, 2019-06-09:
Should be fixed by commit.

closed 2019-06-09 by kasanovic